EverythingSmartHome/everything-presence-one #3: EP1 as bluetooth proxy for home assistant?

amaisano, 2022-12-21:
Does the hardware include bluetooth w/active scanning? I was thinking of also using this wonderful device as a BT proxy for some of my home assistant needs.

Is BT enabled?

amaisano, 2022-12-21:
Tried adding this to the HA Add-on config:

https://www.home-assistant.io/integrations/bluetooth#esphome-requirements

```yaml
esp32_ble_tracker:
bluetooth_proxy:
```

```
Failed config

esp32_improv: [source /config/esphome/.esphome/packages/f43db209/everything-presence-one.yaml:33]
  
  Component esp32_improv cannot be used together with component esp32_ble_tracker.
  authorizer: none
esp32_ble_server: None
  {}
esp32_ble: None
  {}
Component esp32_ble_server cannot be used together with component esp32_ble_tracker

Component esp32_ble cannot be used together with component esp32_ble_tracker
```

I am very new to ESP32 (this is my first device) so I think I just need some guidance.

EverythingSmartHome, 2022-12-21:
Hello!

BT tracking and BT Proxy is disabled by default because it conflicts currently with Improv (which is used for initial setup) - in the future it should be possible to have them on at the same time but currently it meant we had to disable them to allow for easy setup.

They can of course be added after setup. You will need to firstly create your own ESPHome config if you haven't already done so - you can copy the full EP1 config and paste it in using the config from here: https://github.com/EverythingSmartHome/everything-presence-one/blob/main/everything-presence-one.yaml

Replace the WiFi section of this config which currently looks like:

```
wifi:
  ap {}
```

with this:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

You will also need to remove the section that says:
```
esp32_improv:
  authorizer: none
```

And add the following lines back in:
```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

Finally make sure there are no errors, save the config and then install the new config to the EP1.

amaisano, 2022-12-21:
SO, something like this?

https://github.com/EverythingSmartHome/everything-presence-one/compare/main...amaisano:everything-presence-one:patch-1

And then I just use the HA Add-On dashboard and click "Save" and then "Install" with those file contents?

I'm so afraid of "breaking" the EP1 functionality.

EverythingSmartHome, 2022-12-22:
> SO, something like this?
> 
> [main...amaisano:everything-presence-one:patch-1](https://github.com/EverythingSmartHome/everything-presence-one/compare/main...amaisano:everything-presence-one:patch-1)
> 
> And then I just use the HA Add-On dashboard and click "Save" and then "Install" with those file contents?
> 
> I'm so afraid of "breaking" the EP1 functionality.

Looks good to me!

You can't really break the functionality - if you make a mistake then you can simply use the flash page here to restore to factory firmware:

https://everythingsmarthome.github.io/everything-presence-one/flashing.html

If you get stuck then feel free to join the Discord which might be easier to talk you through.

amaisano, 2022-12-28:
RESOLVED

The proposed patch (a combination of the original configuration from EP1 and some additional changes for BT proxy support) works perfectly, but getting it into the EP1 needed some alternative/additional steps.

From the man @EverythingSmartHome himself:

- Since you don't have HTTPS enabled you'll have to do a bit more work
- Safe mode ON anytime you want to do an OTA update (adjusting config/settings wirelessly)
- [Skip installing when adopting things in Add-On]
- [In HA ESP Home Add-On] press install then manual download and choose modern
- Download the file it gives
- Then go to web.esphome.io and install the firmware using that website (over USB)

closed 2022-12-28 by amaisano

nstrelow, 2023-01-04:
Just found this, because I was wondering if the ble_proxy was enabled by default and couldn't find it in the config.

@EverythingSmartHome 
Might by worth sticking this or some explanation on why this is not added by default, with this issue are a small guide on how to enable it.
Since bluetooth proxy being an option is advertised on the shop (and also on the board, love the markings. USB C baby 🥳 )

dhuddle, 2023-01-07:
Has anyone tested this for a bit? I’m setting up my EP1’s now and I’m interested in turning on the proxy function, but as I’ve done this on another (different) esphome device I had and I found that it caused that device to behave a bit strangely.

nstrelow, 2023-01-08:
No idea how it should behave weirdly. If bluetooth is not used for anything else. Power consumption might go up.

amaisano, 2023-01-10:
Mine IS actually behaving weirdly. But I cannot be sure if it is from before or after the proxy config setup. It is constantly disconnecting/becoming unknown/unavailble in HA.

[image]

EverythingSmartHome, 2023-01-10:
> Mine IS actually behaving weirdly. But I cannot be sure if it is from before or after the proxy config setup. It is constantly disconnecting/becoming unknown/unavailble in HA.
> 
> [image]

These seems to happen on a couple users WiFi for some reason - the fix we found is to either update to ESPHome beta 2023.1 or to adjust the power saving mode in the config to "none":

https://esphome.io/components/wifi.html#wifi-power-save-mode

It does seem to be Bluetooth related and one user mentioned 2023.1 beta does fix it for him.

amaisano, 2023-01-15:
> adjust the power saving mode in the config to "none":

  power_save_mode NONE is incompatible with esp32_ble_tracker. Please remove the power save mode. See also https://github.com/esphome/issues/issues/2141#issuecomment-865688582.