EverythingSmartHome/everything-presence-one #6: Entity Naming

macklenc, 2022-12-26:
First off, thanks for the awesome sensor pod! I've run into a bit of an annoyance with the naming of the entities, and I'm struggling to find a robust way to fix it. I have several EP1's, and they all expose e.g. `sensor.humidity`. When I add a second device it becomes `sensor.humidity_1`. Do you have a recommended way to name these entities based on the device name? E.g. if I have one named `ep1_office` I'd like the entities to become e.g. `sensor.ep1_office_humidity`. I've looked into `hass-cli` for bulk renaming the roughly 100 entities I have, but it would require writing a python script to be able to cross reference the device IDs to the entities.

Any pointers would be appreciated!

EverythingSmartHome, 2023-01-02:
There is the "room" variable you can add which should help. First delete the EP1 from Home Assistant to clear away any entities

Open up the ESPHome adopted config an under the ```substitution:``` section, add the following:

```
  room: "Office"
```

Then save, upload the new config and re-add to HA. All the devices should have "Office" prefixed to the entity name

ChocolateNinja, 2023-01-04:
Is is possible to add the room prefix option to all the generated entities? Just to prevent any overlap with multiple EP1's, currently it looks like only the sensors have it.

EverythingSmartHome, 2023-01-04:
> Is is possible to add the room prefix option to all the generated entities? Just to prevent any overlap with multiple EP1's, currently it looks like only the sensors have it.

Yeah it sure is - I will do that at some point soon.

pshempel, 2023-02-08:
I also agree with this, I have 5 of these and it makes writing automation via the GUI, very painful. This leads me to have to write it via YAML so I don't make mistakes in picking the device.  
Once the rest of my devices get here, I will have close to 12 of these devices and it will become unwieldy. I have had to manually rename the devices via ESPhome, but all of the logs and device pickers show the entity name, not the alias that ESPHome creates when renaming the device

macklenc, 2023-07-11:
> I face the issue of HA changing the entities' names when updating EPSHome.
> 
> Having just one EP sensor integrated I find myself with an unavailable e.g. "_binary_sensor.presence_01_pir_" but with a new entity called "_binary_sensor.presence_01_pir_2_". This happens for all its entities.
> 
> How can I avoid this from happening once I update ESPHome and its sensors?

Looks like I don't have this issue, by setting the `room` substitution <most> of my sensors are given a name like `sensor.office_temperature`, though several sensors don't seem to get this substitution like e.g. `status_led`, `mmwave_distance`, `mmwave_led`, `mmwave_on_latency`, `mmwave_off_latency`, `mmwave_sensitivity`, and `mmwave_sensor`. However, none of those sensors are incrementing with updates, here's my config for reference:

```yaml
substitutions:
  name: ep1-office
  temperature_offset: "-6.5"
  room: "My Office"
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: <Key>
```

Zwo12, 2023-07-12:
> Something appears to have changed in ESPHome 2023.6.5 - I have the "room" prefix set up which was working across upgrades up to ESPHome 2023.6.4, but in 2023.6.5 it seems to have been ignored and all my EP1 entities have "Entity Presence One" as the prefix in the friendly names, and "everything_presence_one" as a prefix in the entity ids, for example "binary_sensor.everything_presence_one_occupancy".

Same here… followed the above instruction several times, since I thought the problem was sitting in front of the screen, but always the same result ignoring the „room“ prefix

brofeelgood, 2023-07-13:
As discussed in #80, the "room" substitution has been superseded by "friendly_name". A few of the original entities have been disabled and/or renamed to something else.

Here's an example of the EP1 yaml with the latest substitutions:
https://github.com/EverythingSmartHome/everything-presence-one/blob/main/everything-presence-one.yaml

closed 2024-07-19 by EverythingSmartHome