EverythingSmartHome/everything-presence-one #8: Temperature offset not working

nsleigh, 2023-01-02:
First off thanks for producing this device, it is exactly what I need to control things in my home office. (Apologies if this is not the place for support.)

I am trying to get the temperature a bit more accurate using the offset but it doesn't seem to work (the docs say this is in the advance section but I couldn't find it so I might have missed something). All I have done so far is add this to the existing settings and installed it using ESPHome (over air connection).

substitutions:
  temperature_offset: "-3"

I have downloaded the yaml and I can see it is still set.

Any suggestions?

EverythingSmartHome, 2023-01-02:
> 

Sorry I haven't gotten around to adding details to the advanced section yet but yes you are setting the offset correctly. The default is already -3 so it seems like you are replacing it with the same value that is already present, did you try setting it to something else?

neilsleightholm-paxton, 2023-01-02:
No thought the yaml was a sample not default so will try another value.

nsleigh, 2023-01-03:
Yep that was the issue - changed to -6 and it is more accurately reflecting the ambient temperature.

closed 2023-01-03 by nsleigh