EverythingSmartHome/everything-presence-one #9: WiFi unreliable

cmsj, 2023-01-02:
I'm not sure if this is the same thing as #1 but I'm having a bad time getting my sensor setup. I was getting disconnects even when connecting to its wifi network to choose the SSID, and once I eventually got through that, I'm seeing really terrible performance on my main network:

```
64 bytes from 10.0.88.167: icmp_seq=5 ttl=255 time=3904.011 ms
64 bytes from 10.0.88.167: icmp_seq=6 ttl=255 time=2905.910 ms
64 bytes from 10.0.88.167: icmp_seq=7 ttl=255 time=1904.328 ms
64 bytes from 10.0.88.167: icmp_seq=8 ttl=255 time=901.799 ms
64 bytes from 10.0.88.167: icmp_seq=9 ttl=255 time=5.123 ms
Request timeout for icmp_seq 13
Request timeout for icmp_seq 14
64 bytes from 10.0.88.167: icmp_seq=12 ttl=255 time=3049.849 ms
64 bytes from 10.0.88.167: icmp_seq=13 ttl=255 time=2099.600 ms
64 bytes from 10.0.88.167: icmp_seq=14 ttl=255 time=1095.164 ms
64 bytes from 10.0.88.167: icmp_seq=15 ttl=255 time=92.496 ms
64 bytes from 10.0.88.167: icmp_seq=16 ttl=255 time=1465.333 ms
```

My network is UniFi (UAP-AC-AP-Pro access points and a UDM SE controller), I'm about 3-4 metres from the AP the sensor is connected to, and when I did the initial setup I had the sensor next to the computer that was connecting to it.

cmsj, 2023-01-02:
[logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/10332696/logs.txt)

EverythingSmartHome, 2023-01-02:
> I'm not sure if this is the same thing as #1 but I'm having a bad time getting my sensor setup. I was getting disconnects even when connecting to its wifi network to choose the SSID, and once I eventually got through that, I'm seeing really terrible performance on my main network:
> 
> ```
> 64 bytes from 10.0.88.167: icmp_seq=5 ttl=255 time=3904.011 ms
> 64 bytes from 10.0.88.167: icmp_seq=6 ttl=255 time=2905.910 ms
> 64 bytes from 10.0.88.167: icmp_seq=7 ttl=255 time=1904.328 ms
> 64 bytes from 10.0.88.167: icmp_seq=8 ttl=255 time=901.799 ms
> 64 bytes from 10.0.88.167: icmp_seq=9 ttl=255 time=5.123 ms
> Request timeout for icmp_seq 13
> Request timeout for icmp_seq 14
> 64 bytes from 10.0.88.167: icmp_seq=12 ttl=255 time=3049.849 ms
> 64 bytes from 10.0.88.167: icmp_seq=13 ttl=255 time=2099.600 ms
> 64 bytes from 10.0.88.167: icmp_seq=14 ttl=255 time=1095.164 ms
> 64 bytes from 10.0.88.167: icmp_seq=15 ttl=255 time=92.496 ms
> 64 bytes from 10.0.88.167: icmp_seq=16 ttl=255 time=1465.333 ms
> ```
> 
> My network is UniFi (UAP-AC-AP-Pro access points and a UDM SE controller), I'm about 3-4 metres from the AP the sensor is connected to, and when I did the initial setup I had the sensor next to the computer that was connecting to it.
> 
> Edit: I did an update using web.esphome.io. I think it updated correctly, but it doesn't seem to be helping. `[I][app:102]: ESPHome version 2022.12.3 compiled on Jan 2 2023, 16:25:05`

Hello!

Your logs look OK on quick inspection - it kind of sounds like your power supply isn't performing too well. Can you try another known good power supply and/or plug it into a laptop or computer and see how it is there?

cmsj, 2023-01-02:
Weird, I was seeing that behaviour on a dual port USB-C GaN charger and on a Mac Pro's USB port, but I just moved it over to an Anker multi-port and now the pings seem fine. Glad it was something simple, thanks!

closed 2023-01-02 by cmsj