EverythingSmartHome/everything-presence-one #11: Right way to do updates?

Nuuki9, 2023-01-03:
I got the sensor installed and working. I've had some issues with the sensor crashing howeever, and I see others have resolved that by updating the firmware. However at that point I get a bit stuck.
"
I've tried using the Update button in the sensor's webUI - not specifying a file. It seemed to do something, and if I hit the Update button again it says "Update successful", but its still reporting 2022.11 in the decice screen in HA so I'm not confident that's worked. I'm still getting the sensor locking up.

Others have referenced updating via USB, but I'm not sure how to actually do that. I have ESPHome already running (in Docker) from some testing I did last year, but I've not done much with it. If I fire that up I can onboard a new device, but not sure if that's what I should be doing, as usually that would build new custom firmware, no?

Any chance of a quick summary on the right way to do both OTA and USB based updates?

EverythingSmartHome, 2023-01-03:
To update the initial firmware you will need to use USB - you can do that by just plugging in the EP1 to your computer with a USB-C cable, then heading to the update page in the user guide (make sure to use Chrome for this as Safari and others don't support it): https://everythingsmarthome.github.io/everything-presence-one/flashing.html

Hit the connect button, select the USB-Serial Port from the menu and then hit install. I'd recommend then going back in and removing the device from Home Assistant and re-adding it. It's not required and will work fine but this will clear up the entities for you so they are shown correctly which were fixed in other updates.

Nuuki9, 2023-01-03:
Many thanks! I did have another look in the manual and reaised by browser wasn't supproted, so wasn't seeing the Connect option - tried again and got it all upgraded.

Its all re-added so I'll leave it to see if its now more stable. One slightly odd change is that I can no longer connect to the ESPHome directly via a browser. Its definitely on the same IP and communincating with HA fine. I can aee port 6053 is open but no web server on port 80, so not sure if there was a change or if its an issue at my end? Not a huge issue, but it has been good to be able to reset it when its locked up.

EverythingSmartHome, 2023-01-03:
> Many thanks! I did have another look in the manual and reaised by browser wasn't supproted, so wasn't seeing the Connect option - tried again and got it all upgraded.
>
> Its all re-added so I'll leave [...]

Glad to hear it! The web server component was removed as that was the component that locks up the device, you shouldn't see any more locking up/disconnects with it removed!

Nuuki9, 2023-01-04:
> Glad to hear it! The web server component was removed as that was the component that locks up the device, you shouldn't see any more locking up/disconnects with it removed!

Makes sense. 

So if I wanted to do OTA updates, is that something I would need to do via the ESPHome docker container? I'm not seeing an easy way to add the sensor (allowing me to manage it) without wiping it in the process.

Given how fiddly it can be to get mmwave sensors positioned and configured for maximum performance it would be _ideal_ not to have to take them down every time they needed an update, though if that's the best route its hardly a show stopper.

Thanks for your work on these - getting excellent results compared to my old PIR - just wish I'd not forgotten to order the case when I picked one up :-)

EverythingSmartHome, 2023-01-04:
> Makes sense.
>
> So if I wanted to do OTA updates, is that something I would need to do via the ESPHome docker container? I'm not seeing an easy way to add the sensor (allowing me to manage it) without [...]
OTA updates are done via the ESPHome dashboard yes - it might ask you to "install" the initial configuration when you create the config but you can just skip that. It will import the EP1 config into your dashboard and then you can do OTA updates. Note to do OTA, you need to hit the safe mode button in Home Assistant for this device first, then start the OTA process.

> 
> Thanks for your work on these - getting excellent results compared to my old PIR - just wish I'd not forgotten to order the case when I picked one up :-)
Really happy to hear you are enjoying it! If you have a 3D printer then the STL is available on the shop page if you want to print yourself!