EverythingSmartHome/everything-presence-one #13: Support for external ESPHome installs

mikey0000, 2023-01-06:
Hi, I keep getting errors with ESPHome 

```
INFO Reading configuration /config/everything-presence-one-e01da0.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

dashboard_import: [source /config/.esphome/packages/0c0347a4/everything-presence-one.yaml:46]
  package_import_url: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
  
  [import_full_config] is an invalid option for [dashboard_import]. Please check the indentation.
  import_full_config: False
```

mikey0000, 2023-01-06:
Actually looks to be a bug in an older version of ESPHome pre 2022.12 updated the docker image and now its fine.

closed 2023-01-06 by mikey0000

gukr1969, 2023-11-18:
> Actually looks to be a bug in an older version of ESPHome pre 2022.12 updated the docker image and now its fine.

Hello mikey0000, I had the same problem yesterday when installing EP1. Unfortunately, I am more of a user than a Docker specialist. Could you explain the solution to this problem in more detail so that I can reproduce it step by step. Thank you.