EverythingSmartHome/everything-presence-one #14: EP1 not adding to ESPHome Dashboard

weyrleader, 2023-01-15:
Hi, 

I've added two EP1's to my HA,, they both were automatically discovered by mDNS and added to via ESPHome integration and I can see both devices and their associated sensors, so all good from that perspective. However, neither device was added to the ESPHome dashboard, the documentation says to install the addon, but that was already installed (I've used it for some time). I tried deleting both EP1 devices from the Intergrations page and then readding them both manually, which added them back ok but still not showing in the dashboard. Any ideas?

James

P.S. The main reason I want this it so I can change the code to calibrate the humidity sensor, I've got 3 separate humidity sensors in the same room and they are all wildly different from each other.

EverythingSmartHome, 2023-01-17:
Hello! I'm not sure why they wouldn't be adopting automatically, especially if they are auto discovered in Home Assistant.

To get around this you can simply copy the full config from the GitHub configuration and manually create your own config. Link to the code is [here](https://github.com/EverythingSmartHome/everything-presence-one/blob/main/everything-presence-one.yaml)

weyrleader, 2023-01-17:
Thanks will give that a go tonight and see how I go.

closed 2023-02-24 by EverythingSmartHome