EverythingSmartHome/everything-presence-one #15: Component esp32_improv cannot be used together with component esp32_ble_tracker

DutchDeffy, 2023-01-16:
Hello,

Thanks for this awesome project. I've been having lots of fun while tinkering with it. 

On the webshop it says:

"The ESP32 also provides Bluetooth tracking capabilities, along with the brand new [Bluetooth Proxy] feature in Home Assistant and ESPHome - this allows for extending Home Assistant's Bluetooth range by using remote ESP32's around your house - a function that is a match made in heaven for this board!"

However I get a compiling error when I try to configure the Bluetooth Proxy using the ESPHome version 2022.12.3 and the [recommended configuration](https://esphome.io/components/bluetooth_proxy#complete-sample-recommended-configuration). So adding these lines to the default config.

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

Gives the following error while compiling.

```
Failed config

esp32_improv: [source /config/esphome/.esphome/packages/f43db209/everything-presence-one.yaml:43]
  
  Component esp32_improv cannot be used together with component esp32_ble_tracker.
  authorizer: none
esp32_ble_server: None
  {}
esp32_ble: None
  {}
Component esp32_ble_server cannot be used together with component esp32_ble_tracker

Component esp32_ble cannot be used together with component esp32_ble_tracker
```

Should this be changed in the configuration file hosted on this GitHub? Or should I, as a user, make a copy of this GitHub config and adept these changes to add the bluetooth proxy functionality?

EverythingSmartHome, 2023-01-17:
Glad you are enjoying it :)

The lines you have added are correct, but as the error says, they cannot be active at the same time as another component.

Simply remove the lines:

```
esp32_improv:
  authorizer: none
```

To allow it to work.

The changes cannot be made in the stock firmware as the improv component is used for initial setup, which is why bluetooth proxy is not included in the stock firmware - we would need for ESPHome to allow both components to be active at the same time before I could make the stock firmware include bluetooth proxy. Therefor, you will need to create your own custom configuration

ThreeFN, 2023-01-27:
Forgive me for being pretty rubbish at esphome & yaml, but are there any syntactical tricks that could be used to 'disable' the esp32_improv import while still retaining the rest configuration as it's pulled from the remote package?  I'd rather not loose the 'auto update' by using the remote package pull from this guthub as would happen once I copying the yaml config to my own esphome dashboard and remove improv/add proxy.

For the handful of devices I have that need unique configs (eg some devices have 'use_flash enabled' because I do want them to retain state after a power outage) I have made use of 'package over-writing-substitutions', love to use a trick like that in this case to disable esp32_improv and enable BLE.

Or heaven forbid, provide a spare repo/package that is the proxy version, so you can sub in the github link with the 'BLE version' once you get configured enough to have the device enrolled in esphome?

All this is purely selfish, I'd just like to not to have to come to github every so often to see if I have to copy/pastes a new config when remote packages can do it for me.  Hope it doesn't hurt to ask...

jesserockz, 2023-02-02:
ESPHome 2023.2.0 will allow `esp32_improv` and `bluetooth_proxy` at the same time.

GuySie, 2023-02-18:
> ESPHome 2023.2.0 will allow `esp32_improv` and `bluetooth_proxy` at the same time.

Does that mean, considering 2023.2 is now out, that the stock firmware can now support the bt proxy out-of-the-box?

neildsb, 2023-02-28:
I managed to configure and compile the YAML - upload over USB, I just do not have any Bluetooth devices to verify :-(

Home Assistant 2023.2.5
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230202.0 - latest

Everything_Presence_One
by Everything_Smart_Technology
Firmware: 2023.2.4 (Feb 28 2023, 14:36:24)
Hardware: 1.1.3

[15:00:19][C][bluetooth_proxy:065]: Bluetooth Proxy:
[15:00:19][C][bluetooth_proxy:066]:   Active: YES
[15:00:19][C][esp32_ble:214]: ESP32 BLE:
[15:00:19][C][esp32_ble_server:161]: ESP32 BLE Server:
[15:00:20][C][esp32_improv.component:215]: ESP32 Improv:
[15:00:20][C][esp32_improv.component:217]:   Status Indicator: 'NO'
[15:00:20][C][esp32_ble_tracker:583]: BLE Tracker:
[15:00:20][C][esp32_ble_tracker:584]:   Scan Duration: 300 s
[15:00:20][C][esp32_ble_tracker:585]:   Scan Interval: 1100.0 ms
[15:00:20][C][esp32_ble_tracker:586]:   Scan Window: 1100.0 ms
[15:00:20][C][esp32_ble_tracker:587]:   Scan Type: ACTIVE
[15:00:20][C][esp32_ble_tracker:588]:   Continuous Scanning: True

GuySie, 2023-02-28:
I've tested it last week and so far it compiles and installs but it's been very buggy. Sensor drops out entirely at times. Does not seem to be an issue between esp32 improv and Bluetooth proxy, removing improv does not matter. Removing the ble tracker settings seems to improve the sensor dropout, but means that passive ble messages (like from miflora sensors) become erratic.

DutchDeffy, 2023-04-19:
Closing because the compilation error with both `esp32_improv` and `bluetooth_proxy` doesn't occur anymore since ESPHome version 2023.2. The (passive) Bluetooth tracking compatibility is "possible". Although, as of right now, it might not be that stable. This already great device will only get better along the way!

closed 2023-04-19 by DutchDeffy