EverythingSmartHome/everything-presence-one #16: mmWave led turning back on (randomly?)

poulti, 2023-02-02:
Hello,

Great product, love the combined sensors in one box.
I have 3x EP1s in different rooms. I've turned all leds off (including mmWave) for all of them.
For some reasons, the mmWave led turns back on again randomly (didn't find the pattern/trigger yet). 
It doesn't happen to all 3 at the same time - but it happened to all 3 over time (hope it makes sense).
Any idea what it could come from? Any other idea to force it off?
Thanks

EverythingSmartHome, 2023-02-03:
Hello!

Thanks! Have you got a screenshot of the history of the LED coming back on? Perhaps there is some sort of pattern to it?

Could it be an automation you have that turns on all lights in an area or something?

poulti, 2023-02-03:
Hello,
Thanks for replying. 
I noticed when the led is back on (on the device) the mmWave led switch is still in "off" state. I have to tap the "on" lightning bolt and "off" again to turn it off. The history seems consistency as there is no visible "on" timeslots.
[image]
I'll take note of the next event date/time to pin point it in the history, if that shows something.

2 out of 3 EP1s don't have any automation attached (yet), I just use them for the temp/hum/light sensors for now. But all 3 are affected (at different times)

helicopterrun, 2023-02-04:
I also experienced this issue, but sure of the root cause. Since I don't need the lights for anything I just disabled them in the .yaml file.

poulti, 2023-02-04:
Could you please detail what you changed in the config file?
Thanks

helicopterrun, 2023-02-05:
Assuming you have the ESP Home add-on installed. You should be able to edit the "everything-presence-one.yaml" remove the "light:" code block as well as remove the led section from the "switch:" code block as shown:
```yaml
##### remove this block #####
#############################
#light:
#  - platform: status_led
#    name: "ESP32 Status LED"
#    pin: GPIO32
#    entity_category: config
#############################
## leave this block including "switch:" ##
switch:
  - platform: template
    name: "mmWave Sensor"
    id: "mmwave_sensor"
    entity_category: config
    optimistic: true
    restore_state: true
    turn_on_action:
      - uart.write: "sensorStart"
      - delay: 1s
    turn_off_action:
      - uart.write: "sensorStop"
      - delay: 1s

##### remove this block ##############
######################################
#  - platform: template
#    name: "mmWave LED"
#    id: "mmwave_led"
#    entity_category: config
#    optimistic: true
#    restore_state: true
#    turn_on_action:
#      - switch.turn_off: mmwave_sensor
#      - delay: 1s
#      - uart.write: "setLedMode 1 0"
#      - delay: 1s
#      - uart.write: "saveConfig"
#      - delay: 3s
#      - switch.turn_on: mmwave_sensor
#    turn_off_action:
#      - switch.turn_off: mmwave_sensor
#      - delay: 1s
#      - uart.write: "setLedMode 1 1"
#      - delay: 1s
#      - uart.write: "saveConfig"
#      - delay: 3s
#      - switch.turn_on: mmwave_sensor
######################################
end
```
Leave the rest of the code unchanged, validate and install the code.

poulti, 2023-02-05:
Thanks.
I'm interested to understand what triggered this strange behaviour, but in the meantime, I'll deactivate the mmWave LED.

pshempel, 2023-02-08:
I have the same issue. I have 5 of these and any one of them will randomly turn on. 
Also, the OP seems to not be running the most recent version of the firmware, just FYI

poulti, 2023-02-11:
Updated to latest config + commented the mmWave switch part - still lights up from time to time.
It just lit up now, and in the dashboard/log I see "mmWave sensor turned on". Could that turn on the LED on start?
[image]

EverythingSmartHome, 2023-02-24:
Ah yes, if you toggle the mmWave Sensor on and off using an automation or the button on the dashboard, then it will toggle the LED back on - I looked into this in the past and couldn't see an easy way of retaining the state unfortunately.

BlipRanger, 2023-03-09:
Weirdly, I think I had this same issue occur, but not wit the led state, but the value of the "mmwave_on_latency". Despite it not changing in the HA settings, it clearly had reset to 0 as it was suddenly getting a bunch of false positives. I changed it to 0 in the settings and then back to 2s and it was perfectly fine again.

closed 2024-07-19 by EverythingSmartHome