EverythingSmartHome/everything-presence-one #17: mmWave module failure

pshempel, 2023-02-08:
I have 5 EP1, of which one has a mmWave module that will not show activity, I have moved it between different devices and got the same results. 
Now that the new version of the firmware no longer exposes the factory reset mmWave firmware, I don't have any way to see if that will do anything to resolve this.

EverythingSmartHome, 2023-02-08:
Hi,

Sorry to hear that - is it possible the mmwave was accidently connected to the voltage pins the wrong way around? Do you have a voltmeter to measure if the device has power or not?

You can enable the factory reset button by adding the following line to the ESPHome substitution section in your ESPHome dashboard:

```
factory_reset_disabled: "false"
```

If you are still having trouble, please contact support using the website and we will get you sorted one way or another

Thanks,
Lewis

euDominic, 2023-03-08:
Hi Lewis.

Sorry to bother you here, but no one has responded to my email for 2 days.
I have also already tried contacting them via the homepage, but no response.

I also have a problem with one of my total 7 mmWave sensors.
No matter in which of the 7 boards I plug the sensor, it permanently shows a presence.
I have reflashed each board with this one sensor x times via USB cable, also activated the factory reset of the sensor in the .yml and tried this, without success.
All other sensors always go cleanly into a clear state under the same conditions and settings.
Reducing the sensitivity and / increase the on latency also brought no improvement.
Therefore, I must assume that the sensor has a fundamental problem that cannot be solved by me without a replacement.
I would ask for your help.

Thanks for the support

closed 2023-07-27 by EverythingSmartHome