EverythingSmartHome/everything-presence-one #18: Firmware Update 2023.2.3 won't install

TomFrank19087, 2023-02-23:
When trying to install the 2023.2.3 Firmware update, it says it's installing but when it 'finishes' it just offers to install the update again and shows the old version as installed. Watching the log in ESPHome, the OTA download starts after 2 or 3 minutes but never gets past about 10% before it seems to time out. What am I doing wrong? Thanks in advance!

EverythingSmartHome, 2023-02-23:
Make sure to enable safe mode on the device before attempting to OTA - you can find the button on the device page in Home Assistant

TomFrank19087, 2023-02-23:
Thanks for the tip, but I can't find the safe mode switch anywhere. The device info in the docs don't show it either. Obviously I'm looking in the wrong place. The page I'm looking at I can get to from either the EP1 entry on the devices page or from '1 device' for EP1 on the integrations page.

EverythingSmartHome, 2023-02-23:
This image should help: 
[image]

jarnonline, 2023-02-23:
I have the same problem, but no save mode button
[image]

EverythingSmartHome, 2023-02-23:
@jarnonline That is an old version, make sure to update with the cable first to enable the safe mode button (and other options)

jarnonline, 2023-02-23:
it worked thanks!

EverythingSmartHome, 2023-02-23:
@TomFrank19087 The user guide has the update page: https://everythingsmarthome.github.io/everything-presence-one/flashing.html

TomFrank19087, 2023-02-23:
I was able to find that page on my own. After switching to Chrome, did the cable update without a hitch. Then the firmware updated as it was supposed to. Many thanks!