EverythingSmartHome/everything-presence-one #19: mmWave False Positives

Nuuki9, 2023-02-24:
I've been using a PresenceOne for a couple of months with great success - being able to set the detect distance for mmwave has been a game changer in handling accurate room presence, without "bleeding" over into other rooms when doors are open.

However, recently I'm seeing frequent false positives on the mmWave sensor - around 5 an hour in the middle of the night. Now I realise that these sensors need some adjustments, but I had already done that and had it working perfectly. Here's a quick summary of the setup though.

The sensor is facing into a medium sized room with no windows in its LOS. The distance is set to be within the room only. I'm getting these false positive in the middle of the night when everyone is in bed, and we have no pets. No ceiling fans or other things designed to move.

I have tried turning the sensitivity down, but even at 4 I'm still getting them. The PIR isn't triggering, plus I have a different mmWave sensor in a different part of what are adjoining areas, and that's not triggering either.

The only change is that I recently put the sensor into its case - when I originally ordered it I stupidly missed the case off, so it was working without for a while. I've checked that the firmware is relatively up to date (2022.12.3), though haven't checked today.

So now I'm stumped. As I write this I realise that an obvious test would be to face the sensor up against a wall, as that would eliminate any question of it triggering based on actual movement. I'll do that now and report back, but in the meantime does anyone have any ideas?

Achronite, 2023-02-24:
Mine did the same, so I now have it set to trigger based on PIR only and switch off when occupancy goes 'off'.  This does the trick for me.

Nuuki9, 2023-02-24:
> Mine did the same, so I now have it set to trigger based on PIR only and switch off when occupancy goes 'off'. This does the trick for me.

Thanks - the PIR is working fine, but its the mmWave that makes this sensor useful. The fact that its triggering when facing up against a wall seems to be pretty conclusive that its faulty, so I'll see about a replacement.

EverythingSmartHome, 2023-02-24:
> Thanks - the PIR is working fine, but its the mmWave that makes this sensor useful. The fact that its triggering when facing up against a wall seems to be pretty conclusive that its faulty, so I'll [...]

Placing them against a wall isn't a great test actually because the signal can easily bounce of walls according to the distance parameter. It's unlikely that the unit is faulty. The fact that you went from 5 an hour when facing into the room to 19 an hour when pointing at the wall would seem to support this ;)

Generally people just don't realise how tiny a movement this unit can detect and they generally do have something moving in the room and it's about trying to track down what is moving.

However, the "mmWave On Latency" setting was added in firmware 1.1.3 to help minimise the sensor being too sensitive, it's the exact purpose of this setting. Add 0.5s of on latency to the mmwave and you should see a big reduction, if not eliminated (assuming you just don't have something big moving in the room)

Nuuki9, 2023-02-24:
Thanks for the info. I've added in 0.5s of latency and will point it back into the room.

Point taken regarding sensitivity. Its only that it was working rock solidly that makes me feel something's amiss now. I have also updated to the very latest firmware.

Anyway I'll leave it overnight and see how it does.

EverythingSmartHome, 2023-02-24:
Generally the 0.5s of on latency really cleans things up a lot for people who have "noisier" environments, but if need be you can add a little more if you need to. You should see a big difference with 0.5 at least!

EverythingSmartHome, 2023-02-25:
Good to hear!

There are a few things that need to be added that are in the pipeline but other things are taking priority just now, but it will be added sometime soon.