EverythingSmartHome/everything-presence-one #20: Couldn't update / setup the sensor on M1 Mac or Macbook 14 Pro with Mac OS 13 & Chrome

nimmerpeople, 2023-02-28:
Hi,

Received the sensor today and couldn't get it to update on an M1 or a Macbook 14 Pro.
They both "connected" and i got the initial menu pop up - but the install failed on both. Was keen to get it working - so tried my son's windows laptop and that worked first time.

Just mention it here incase others see issues. Happy to troubleshoot further now I know it works. 

thanks for the kit - looks superb. Excited to get it working now.

EverythingSmartHome, 2023-03-01:
Hello!

I believe this is due to the M1 Mac not having a driver for it because of the change of architecture - I had a quick search and see comments here: 

https://github.com/LilyGO/LILYGO-T-OI/issues/3#issuecomment-907645945
https://github.com/WCHSoftGroup/ch34xser_macos
And also:
https://arduino.stackexchange.com/questions/85822/is-there-an-apple-silicon-m1-driver-for-ch34x-devices

I'm not sure if these solutions work or if it's been resolve in MacOS 13 natively as I don't have any hardware to test that on but hopefully that helps.

nimmerpeople, 2023-03-01:
> LilyGO/LILYGO-T-OI#3 (comment)

Yes, I found that last night, so pleased it was explained.

Worked fine on a borrowed windows laptop and all good now.

thanks!