EverythingSmartHome/everything-presence-one #21: Problems with naming in ESPHome

colinnewman2000, 2023-02-28:
Firstly, please excuse my ignorance.

I am trying to change the default names in ESPHome so I can try to match with meaningful names in HA. Whatever way I try to rename the device in ESPHome (via the rename option or using the substitution in YAML, it does a load of stuff then just tells me it can't resolve the name and more and the device then just appears offline. 

Also it seems to keep finding the old name and asking me to adopt it again, but then tells me it doesn't exist. Even if I remove and re-install ESPHome, I seem to need to manually delete all the files before I do so, or all the naff config is still there.

Guess I am just being daft.

EverythingSmartHome, 2023-03-01:
Hello!

You can rename the entities in Home Assistant to whatever you prefer without re-flashing or changing the ESPHome side if you are having a problem with changing the config manually, the result should be the same

hellcry37, 2023-03-31:
Ok so I went into this:

What actually happens is an issue caused by you setting name_add_mac_suffix: true

Now I understand how that suppose to work but just fails badly, at least in my tests:

What you should do is this:

name_add_mac_suffix: false
use example:

  name: "bathroom-epo"
  room: "Bathroom"

This will give you the best results:
sensors like for example:
sensor.bathroom_humidity
sensor.bathroom_illuminance
sensor.bathroom_temperature
sensor.{{ room }}_function -> you get the point

Hope this helps.

closed 2024-07-19 by EverythingSmartHome