EverythingSmartHome/everything-presence-one #22: Issue updating via ESPHome

Pdh362, 2023-03-01:
Issue: Updating via ESPHome running in HA fails with an error, 'ERROR Error sending data: [Errno 104] Connection reset by peer'

Hardware: Wifi 2.4 access point within 12 feet of P1 device, which is active and connected fine. 

Symptoms: Requesting a device update via ESPHome in HA, the device compiles and assembles the bin file successfully and begins uploading, at which point it fails. See attached logs. 

I'm going to try a beefier power supply for the P1 sensor in case this is relevant - any other reasons this could occur?
[logs_update-all.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/10860440/logs_update-all.txt)

EverythingSmartHome, 2023-03-01:
Make sure to enable "Safe mode" in Home Assistant for the device before attempting an OTA. See duplicate: https://github.com/EverythingSmartHome/everything-presence-one/issues/18

Pdh362, 2023-03-01:
Many thanks - all resolved, and apologies I hadn't spotted the use of the Safe Mode

closed 2023-03-01 by Pdh362

EverythingSmartHome, 2023-03-01:
No problems, need to get round to adding to the instructions!!

Glad it worked.