EverythingSmartHome/everything-presence-one #23: EP1 keeps going offline / unavailable

didgeriwho, 2023-03-02:
Even after updating to the latest firmware (firmware: 2023.2.4) per #1, my EP1 keeps going offline. I have 2 of the sensors and both have the same issues of going offline. I've tried switching power adapters and issue still persists. 

ESPHome yaml:
```
substitutions:
  name: everything-presence-one-de99b4
packages:
  Everything_Smart_Technologies.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

I'm seeing stuff in the logs but not sure how to remediate (FYI - I also have 2 HA instances currently running right now)
```
D][bh1750.sensor:159]: ' Illuminance': Got illuminance=5.1lx
[D][sensor:127]: ' Illuminance': Sending state 5.09350 lx with 1 decimals of accuracy
[D][api:102]: Accepted ::FFFF:192.168.1.199
[W][api.connection:083]: Home Assistant 2023.3.0 (::FFFF:192.168.1.199): Connection closed
[D][shtcx:100]: Got temperature=26.60°C humidity=23.92%
[D][sensor:127]: ' Temperature': Sending state 23.60378 °C with 1 decimals of accuracy
[D][sensor:127]: ' Humidity': Sending state 28.92120 % with 1 decimals of accuracy
[D][bh1750.sensor:159]: ' Illuminance': Got illuminance=4.5lx
[D][sensor:127]: ' Illuminance': Sending state 4.52756 lx with 1 decimals of accuracy
[W][api.connection:081]: ESPHome Logs 2023.2.4 (::FFFF:192.168.1.199): Connection reset
[W][api.connection:081]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connection reset
[D][api:102]: Accepted ::FFFF:192.168.1.199
[D][api:102]: Accepted ::FFFF:192.168.1.99
[D][bh1750.sensor:159]: ' Illuminance': Got illuminance=4.8lx
[D][sensor:127]: ' Illuminance': Sending state 4.75394 lx with 1 decimals of accuracy
[D][api:102]: Accepted ::FFFF:192.168.1.99
[W][api.connection:083]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connection closed
[W][api.connection:083]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connection closed
[D][api:102]: Accepted ::FFFF:192.168.1.99
[D][api.connection:918]: Home Assistant 2023.3.0 (::FFFF:192.168.1.199): Connected successfully
[W][api.connection:083]: Home Assistant 2023.3.0 (::FFFF:192.168.1.199): Connection closed
[D][api:102]: Accepted ::FFFF:192.168.1.199
[D][shtcx:100]: Got temperature=26.54°C humidity=24.01%
[D][sensor:127]: ' Temperature': Sending state 23.53969 °C with 1 decimals of accuracy
[D][sensor:127]: ' Humidity': Sending state 29.01123 % with 1 decimals of accuracy
[D][api.connection:918]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connected successfully
[D][api.connection:918]: ESPHome Logs 2023.2.4 (::FFFF:192.168.1.199): Connected successfully
[D][api:102]: Accepted ::FFFF:192.168.1.199
[D][bh1750.sensor:159]: ' Illuminance': Got illuminance=5.1lx
[D][sensor:127]: ' Illuminance': Sending state 5.09350 lx with 1 decimals of accuracy
[W][api.connection:081]: ESPHome Logs 2023.2.4 (::FFFF:192.168.1.199): Connection reset
[W][api.connection:083]: Home Assistant 2023.3.0 (::FFFF:192.168.1.199): Connection closed
[W][api.connection:081]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connection reset
[D][api:102]: Accepted ::FFFF:192.168.1.199
[D][api:102]: Accepted ::FFFF:192.168.1.99
[D][api.connection:918]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connected successfully
[D][api.connection:918]: Home Assistant 2023.3.0 (::FFFF:192.168.1.199): Connected successfully
[D][bh1750.sensor:159]: ' Illuminance': Got illuminance=5.4lx
[D][sensor:127]: ' Illuminance': Sending state 5.43307 lx with 1 decimals of accuracy
[D][esp-idf:000]: E (2433428) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 528 bytes)

[D][esp-idf:000]: E (2433430) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 640 bytes)

[D][esp-idf:000]: E (2433442) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 644 bytes)

[D][esp-idf:000]: E (2433455) MDNS: Cannot allocate memory (line: 216, free heap: 676 bytes)

[D][esp-idf:000]: E (2433467) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 680 bytes)

[D][api.connection:159]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99) requested disconnected
[D][api:102]: Accepted ::FFFF:192.168.1.99
[D][api.connection:918]: Home Assistant 2023.3.0 (::FFFF:192.168.1.99): Connected successfully
[D][shtcx:100]: Got temperature=26.53°C humidity=24.01%
[D][sensor:127]: ' Temperature': Sending state 23.53168 °C with 1 decimals of accuracy
[D][sensor:127]: ' Humidity': Sending state 29.01276 % with 1 decimals of accuracy
[D][bh1750.sensor:159]: ' Illuminance': Got illuminance=5.7lx
[D][sensor:127]: ' Illuminance': Sending state 5.65945 lx with 1 decimals of accuracy
[D][api:102]: Accepted ::FFFF:192.168.1.199
[D][api.connection:918]: ESPHome Logs 2023.2.4 (::FFFF:192.168.1.199): Connected successfully
[D][bh1750.sensor:159]: ' Illuminance': Got illuminance=5.3lx
[D][sensor:127]: ' Illuminance': Sending state 5.31988 lx with 1 decimals of accuracy
```

See below for full logs
[esp-web-tools-logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/10874458/esp-web-tools-logs.txt)

EverythingSmartHome, 2023-03-02:
Can you post the full startup logs from a fresh serial connection? When it's running and you are connected to the serial port, press the "reset device" button and capture the logs before and after for around 30-60 seconds.

Can you also post a full page screenshot from the device info page in Home Assistant?

didgeriwho, 2023-03-03:
Unavailable
[image]

Up and Running
[image]

Also I'm not sure if this is indicative of anything but seems like ESPHome thinks I'm still on 2022.12.3 and that there's an update available even though I'm on the latest firmware.
[image]

Logs
[everything-presence-one-de99b4_logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/10877344/everything-presence-one-de99b4_logs.txt)

EverythingSmartHome, 2023-03-03:
Can you try reflashing the stock official firmware from the user guide? I haven't yet tested 2023.2.4 (although it should be fine)

kristoferericsson, 2023-03-08:
I have the same issue, i believe that it started after i updated to to the latest firmware. before i was on the december firmware release.

EverythingSmartHome, 2023-03-08:
> I have the same issue, i believe that it started after i updated to to the latest firmware. before i was on the december firmware release.

Can you post your config for it?

kristoferericsson, 2023-03-08:
> > I have the same issue, i believe that it started after i updated to to the latest firmware. before i was on the december firmware release.
> 
> Can you post your config for it?

Are you referring to the download diagnostics present in the device info?

/K

EverythingSmartHome, 2023-03-08:
Do you have a config inside ESPHome that you adopted and/or tweaked?

Sure, screenshots of device info would be great too.

kristoferericsson, 2023-03-08:
I i have done no tweaking just adopted the device. It was just yesterday that i installed ESPHome because i wanted to look into the possibility to enable Bluetooth proxy. But directly after i had adopted the device it went offline and showed this behavior of dropping connection from time to time. Now i deleted the devices and uninstalled ESPHome to see if that played any role. Now i have re-flashed one of my two with the stock firmware again

[image]

...after uninstalling ESPHome it seems like the disconnects are not happening... but maybe it to early to tell

EverythingSmartHome, 2023-03-08:
Strange, does seem to be related at least somewhat to newer versions of ESPHome but will need to keep an eye on it. Let me know how it goes!

kristoferericsson, 2023-03-08:
It was to early, both of them are intermittently unavailable

[image]

Gordontgopher, 2023-03-08:
I am also having similar issues and am seeing entries in the HA core log, not sure if they help, but it appears that HA is having issues with all my ESPHome devices;

`Logger: aioesphomeapi.connection
Source: /usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py:577
First occurred: 3 March 2023 at 22:22:35 (199 occurrences)
Last logged: 16:27:11

everything-presence-one-8e7ea0 @ 192.168.1.176: Connection error occurred:
esp32-bluetooth-proxy-149808 @ 192.168.1.61: Connection error occurred:
tagreader @ 192.168.1.26: Connection error occurred:
atom-speaker @ 192.168.1.41: Connection error occurred:
everything-presence-one-8e7ea0 @ 192.168.1.175: Connection error occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 536, in send_message_await_response_complex
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 320, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 645, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 554, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 539, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
`

Additionally;

`Logger: aioesphomeapi.connection
Source: runner.py:179
First occurred: 3 March 2023 at 22:40:14 (677 occurrences)
Last logged: 16:16:14

everything-presence-one-8e7ea0 @ 192.168.1.175: Connection error occurred: [Errno 104] Connection reset by peer
tagreader @ 192.168.1.26: Connection error occurred: EOF received
atom-speaker @ 192.168.1.41: Connection error occurred: [Errno 104] Connection reset by peer
esp32-bluetooth-proxy-149808 @ 192.168.1.61: Connection error occurred: [Errno 104] Connection reset by peer
esp32-bluetooth-proxy-149808 @ 192.168.1.61: Connection error occurred: EOF received`

Is there anything I can do to help debugging? 

I will raise an issue on ESPHome as all my devices are impacted, but wanted to share here in case it gave any pointers!

EverythingSmartHome, 2023-03-08:
Very interestingly, there was an update for Home Assistant just released a few minutes ago, 2023.3.2 - https://github.com/home-assistant/core/releases

Looking at the release notes, there is this item:

Bump aioesphomeapi to 13.4.2 ([@bdraco](https://github.com/bdraco) - https://github.com/home-assistant/core/pull/89210) ([esphome docs](https://www.home-assistant.io/integrations/esphome/)) (dependency)

And checking what that fixes, shows it fixes:

This PR fixes or closes issue: fixes https://github.com/home-assistant/core/issues/85062 fixes https://github.com/home-assistant/core/issues/88535 fixes https://github.com/home-assistant/core/issues/89047

I would say it's certainly worth updating as checking the logs in some of those issues, they look very similar to your logs above.

Gordontgopher, 2023-03-08:
Just saw that update. I'll update to 2023.3.2 and provide an update on the issue.

amaisano, 2023-03-08:
Core HA 2023.3.2 did not work with:
- EP1 Firmware: `2023.1.0-dev (Jan 22 2023, 15:12:58)`
- ESP Home Dev HA Add-On
- bt_proxy config added to EP1 config

amaisano, 2023-03-08:
I'm stable (no disconnects) on:

```
Firmware: 2023.2.4 (Mar 8 2023, 21:09:27)
Hardware: 1.1.3
```

and
- EspHome  2023.2.4
- HA Core 2023.3.2

This is after a fresh factory reset of the EP1 using the guide's "Updating" page: https://everythingsmarthome.github.io/everything-presence-one/updating.html

Dec68, 2023-04-12:
Currently running the following versions 

Add-on version: 2023.3.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 9.5  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.4.2
 Home Assistant Supervisor: 2023.04.0

Unfortunately the entities have disconnected and disappeared from the EP1, only entity showing is the firmware update.

[image]

the device logs show other entity information however these do not arrive at the integration. there is also an error number 11 showing. I have no idea what that means.

[22:27:59][C][api:139]:   Address: everything-presence-one-efce38.local:6053
[22:27:59][C][api:141]:   Using noise encryption: YES
[22:27:59][C][improv_serial:032]: Improv Serial:
[22:28:09][D][bh1750.sensor:159]: ' Illuminance': Got illuminance=114.5lx
[22:28:09][D][sensor:127]: ' Illuminance': Sending state 114.54723 lx with 1 decimals of accuracy
[22:28:11][D][api:102]: Accepted 192.168.1.26
[22:28:11][D][api:102]: Accepted 192.168.1.26
[22:28:11][W][api.connection:071]: 192.168.1.26: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[22:28:11][W][api.connection:071]: 192.168.1.26: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[22:28:39][D][bh1750.sensor:159]: ' Illuminance': Got illuminance=114.0lx
[22:28:39][D][sensor:127]: ' Illuminance': Sending state 113.98129 lx with 1 decimals of accuracy
[22:28:40][D][shtcx:100]: Got temperature=22.53°C humidity=42.79%
[22:28:40][D][sensor:127]: ' Temperature': Sending state 19.52625 °C with 1 decimals of accuracy
[22:28:40][D][sensor:127]: ' Humidity': Sending state 47.79175 % with 1 decimals of accuracy
[22:29:09][D][bh1750.sensor:159]: ' Illuminance': Got illuminance=112.6lx
[22:29:09][D][sensor:127]: ' Illuminance': Sending state 112.62302 lx with 1 decimals of accuracy
[22:29:11][D][api:102]: Accepted 192.168.1.26
[22:29:11][D][api:102]: Accepted 192.168.1.26
[22:29:11][W][api.connection:071]: 192.168.1.26: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[22:29:11][W][api.connection:071]: 192.168.1.26: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[22:29:39][D][bh1750.sensor:159]: ' Illuminance': Got illuminance=114.1lx
[22:29:39][D][sensor:127]: ' Illuminance': Sending state 114.09447 lx with 1 decimals of accuracy
[22:29:40][D][shtcx:100]: Got temperature=22.46°C humidity=42.92%
[22:29:40][D][sensor:127]: ' Temperature': Sending state 19.46483 °C with 1 decimals of accuracy
[22:29:40][D][sensor:127]: ' Humidity': Sending state 47.91687 % with 1 decimals of accuracy

This has only happened in the last number of hours, prior to that all entities were operating perfectly.

how do I fix this, I have reinstalled the firmware and the integration into home assistant with no success. what can I try next?

EverythingSmartHome, 2023-04-12:
> Currently running the following versions
> 
> Add-on version: 2023.3.2 You are running the latest version of this add-on. System: Home Assistant OS 9.5 (amd64 / generic-x86-64) Home Assistant Core: 2023.4.2 Home Assistant Supervisor: 2023.04.0
> 
> Unfortunately the entities have disconnected and disappeared from the EP1, only entity showing is the firmware update.
> 
> [image]
> 
> the device logs show other entity information however these do not arrive at the integration. there is also an error number 11 showing. I have no idea what that means.
> 
> This has only happened in the last number of hours, prior to that all entities were operating perfectly.
> 
> how do I fix this, I have reinstalled the firmware and the integration into home assistant with no success. what can I try next?

This looks like a different issue, did you update firmware or the add-on before it stopped working?

First thing I would try is to delete the EP1 from Home Assistant integration, delete it from ESPHome Add-on, re-install the firmware from the user guide and then connect it back to Home Assistant - do not add it to the ESPHome add-on. See if you get the entities showing correctly then.

Dec68, 2023-04-12:
Funny result, Tried that didn't work, used esp home to rename and ended up with all the entities back, well doubled up but visible. rookie error I guess, some small tweaks with automations and it looks like I am back on the road

closed 2023-07-27 by EverythingSmartHome