EverythingSmartHome/everything-presence-one #25: support for BLE proxy.

alexisiglesias1, 2023-03-05:
Since ESPHome 2023.2.0 currently allow esp32_improv and bluetooth_proxy at the same time, is it possible include this configuration in the stock firmware?, in order to keep the OTA update without using custom configuration

EverythingSmartHome, 2023-03-05:
Not currently, there have been a couple of users trialing it and running into disconnection issues with it enabled - will continue to monitor.

MiChAeLoKGB, 2023-05-18:
Was there any progress with this? I am expecting a few BLE devices and was hoping to use my EP ones as proxies, since they are advertised on the store page as a BT proxy.

EverythingSmartHome, 2023-05-18:
Yes there is quite a few people running the workaround in the Discord without issues AFAIK.

You need to switch to the esp-idf framework in the ESPHome config, then disable captive portal and esp32_improv as those components aren't compatible with esp-idf (but are only needed for initial setup anyways).

Make sure to flash via USB-C cable rather than OTA when switching to esp-idf

martindinicolino, 2023-05-19:
> Yes there is quite a few people running the workaround in the Discord without issues AFAIK.
>
> You need to switch to the esp-idf framework in the ESPHome config, then disable captive portal and [...]

would it be possible to publish a tutorial for newbies?

MiChAeLoKGB, 2023-05-19:
I have checked the discord, just to be sure, and it seems relatively easy to enable, only thing I would probably as of you is to publish a yaml with `captive_portal` and `esp32_improv` disabled, so people do not have to copy the entire config just to comment out those 2 lines.
Either have it as a separate file or on a special BLE beta branch if you are not ready to have it in default FW yet.

A bit OT:
I think using Discord forum for a support of a device like this is a bad idea. Trying to find any actual solution is almost impossible, especially when you try to use Google to search for it.
Discord is nice for a community to discuss things and quickly debug something, but actual issues should be solved on easily searchable place, ideally the same one you get your yaml from (so this GitHub in this case). If you do solve it on Discord due to chat being faster a solution should still be posted to an issue on GitHub.
Btw, you do not link to your Discord anywhere on this GitHub.

EverythingSmartHome, 2023-05-19:
> I have checked the discord, just to be sure, and it seems relatively easy to enable, only thing I would probably ask of you is to publish a yaml with `captive_portal` and `esp32_improv` disabled, so people do not have to copy the entire config just to comment out those 2 lines. Either have it as a separate file or on a special BLE beta branch if you are not ready to have it in default FW yet.
> 
> A bit OT: I think using Discord forum for a support of a device like this is a bad idea. Trying to find any actual solution is almost impossible, especially when you try to use Google to search for it. Discord is nice for a community to discuss things and quickly debug something, but actual issues should be solved on easily searchable place, ideally the same one you get your yaml from (so this GitHub in this case). If you do solve it on Discord due to chat being faster a solution should still be posted to an issue on GitHub. Btw, you do not link to your Discord anywhere on this GitHub.

Its something I have intentionally not published at the moment since it's still under testing from the community, I don't want to publish something where there could still potentially be disconnection issues. Once that changes and testers are happy with the results we can certainly publish a firmware for it with it enabled.

Understand your point with Discord but for example what about people who send emails for support? Those conversations are in no way searchable. Discord at least is a public space and is easily searchable (through Discord). It's also just much faster to resolve issues since it's instant messaging which most people so far seem to appreciate it. If people want support on GitHub then I'm happy to answer here, and if people want support on Discord then I'm happy to answer there.

nagydavid, 2023-08-12:
@EverythingSmartHome May I ask, is there any update on this?

reefland, 2023-10-15:
@EverythingSmartHome - can you please provide a status on BLE Proxy?  Preorder page listed it as a feature...

[image]

Been over a year now, and there is no mention of it within the guide if this is supported or not.

Thanks!

wjdp, 2023-11-18:
Been trying to get this working today. I've taken the current YAML file from this repo and done the following:

```
 esp32:
   board: esp32dev
+  framework:
+    type: esp-idf

- captive_portal:
- 
- improv_serial:
- 
- esp32_improv:
-   authorizer: none

+ bluetooth_proxy:
+   active: true
+ 
+ esp32_ble_tracker:
+   scan_parameters:
+     active: true
+
+ text_sensor:
+   - platform: ble_scanner
+     name: "BLE Devices Scanner"
```

I'm now seeing scan results as a sensor on the device!

marcinwasilewski, 2023-11-19:
Thanks @wjdp, I can confirm that the tracker and proxy work with this config
Will be testing if there are any issues over the next few days

m4rkireland, 2023-12-20:
@marcinwasilewski how did the testing go?

marcinwasilewski, 2023-12-21:
I haven't had any issues so far. Ble works and so does everything else it seems

m4rkireland, 2023-12-21:
@EverythingSmartHome any chance of making a new release with this included?

picotrain77, 2023-12-27:
> @EverythingSmartHome any chance of making a new release with this included?

That would be great as I truly don't understand esp or how to edit to add BLE

marcinwasilewski, 2024-01-20:
I suspect one of the latest esphome releases breaks the esp-idf compatibility. One of my EPO that I updated to 2023.12.8 goes into bootloop, so I had to change esp-idf back to arduino (and remove BLE).

The other one is still at 2023.11.2 and esp-idf works fine as well as the bluetooth proxy

townsmcp, 2024-01-20:
@marcinwasilewski i wonder if it’s possible to downgrade as e back to 2024.11.2?

marcinwasilewski, 2024-01-21:
I'm sorry, it looks like I jumped the gun. The bluetooth proxy does not actually work on the other device which is still on 2023.11.2, and I'm not sure if it ever worked because I always had both devices running. 

Since I do need bluetooth proxy to operate my switchbot curtains, I'm going to investigate how I can downgrade and see if it helps, hopefully it's not a hardware issue.

marcinwasilewski, 2024-01-21:
So I solved the issue, and now my device is no longer bootlooping and BLE works. I didn't downgrade esphome, sourcing the config locally from an earlier commit of this repository solved the issue.
I explained here: https://github.com/EverythingSmartHome/everything-presence-one/issues/141#issuecomment-1902626067

townsmcp, 2024-01-21:
@marcinwasilewski excellent news that you have it working again. So are you on ESPHome 2023.12.8?
Can you also confirm if you did the following (I’m pretty new to ESPHome so don’t really understand how it all works):

1. Browse into ESPHome
2. Click ‘Edit’ on the EPO device you want to fix
3. Edit line 5 FROM `Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-beta.yaml@main` TO `Everything_Smart_Technology.Everything_Presence_One: github://EverythingSmartHome/everything-presence-one/blob/b56b71f2ae924af0ee2145df2d8a4864d0bb702b/everything-presence-one.yaml@main`
4. Save changes
5. Install updates to device and update wirelessly

marcinwasilewski, 2024-01-21:
Yes I'm on ESPHome 2023.12.8
I didn't do the steps you mentioned but I think they should work. I just downloaded the file from this commit, and put it in the esphome folder, and then replaced the github:// link with my local path like this

```
packages:
  Everything_Smart_Technology.Everything_Presence_One: !include everything-presence-one.yaml
```

townsmcp, 2024-01-21:
Thank you @marcinwasilewski that seems to have worked for me as well :)
I have the beta firmware installed on my 3 EPO's so I did the same, but used the beta firmware as the basis

closed 2024-07-19 by EverythingSmartHome