EverythingSmartHome/everything-presence-one #27: mmWave LED constantly flashing

te-hb-whatever, 2023-03-10:
This. It is constantly flashing. Changing the "mmWave LED" switch has no effect whatsoever.

oneseventhree, 2023-05-10:
> Have you tried updating the EP1 via the online installer ? Go here: https://everythingsmarthome.github.io/everything-presence-one/updating.html and check #41 if you have issues

I just got mine and I have updated today and mine is still flashing - any ideas:
[image]

This is the light:
[image]

oneseventhree, 2023-05-11:
> If you are able to join the Discord I can try and help troubleshoot what is going on here

Weirdly enough I woke up this morning and the light was off and is staying off. So I guess it’s ok for now! Thanks :)

closed 2024-07-19 by EverythingSmartHome