EverythingSmartHome/everything-presence-one #29: Manual for "editing ESP home code" kinda "interrupted"...

raphael303, 2023-03-11:
Hi

I have successfully added the sensor to my HA, but I want to edit the code now. Sure I have ESP-home installed and built a couple of my own devices with ESPs. However I am unsure, how I proceed now with the yaml file here on GitHub.

Is there a pretty way, to "transfer" my existing EPO from a "blind" installation as per the manual, to one where it actually shows up in ESP-home and the ability to edit the code?

Or do I just have to reflash it with the yaml here on github?

raphael303, 2023-03-11:
Never mind! It suddenly showed up in ESP home and could be adopted! For some reason it didn't before. All good.

raphael303, 2023-03-11:
Ok, or not.
Log looks like this:

[08:38:08][D][binary_sensor:036]: ' PIR': Sending state OFF
[08:38:12][D][binary_sensor:036]: ' PIR': Sending state ON
[08:38:23][D][binary_sensor:036]: ' PIR': Sending state OFF
[08:38:23][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:23][W][api.connection:071]: ::FFFF:192.168.1.103: Socket operation failed: BAD_INDICATOR errno=11
[08:38:23][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:23][W][api.connection:071]: ::FFFF:192.168.1.103: Socket operation failed: BAD_INDICATOR errno=11
[08:38:26][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:26][W][api.connection:083]: aioesphomeapi (::FFFF:192.168.1.103): Connection closed
[08:38:26][D][api.connection:159]: Home Assistant 2023.2.5 (::FFFF:192.168.1.103) requested disconnected
[08:38:26][D][binary_sensor:036]: ' PIR': Sending state ON
[08:38:26][D][bh1750.sensor:159]: ' Illuminance': Got illuminance=92.1lx
[08:38:26][D][sensor:127]: ' Illuminance': Sending state 92.13582 lx with 1 decimals of accuracy
[08:38:26][D][binary_sensor:036]: ' mmWave': Sending state ON
[08:38:26][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:26][D][api.connection:918]: Home Assistant 2023.2.5 (::FFFF:192.168.1.103): Connected successfully
[08:38:47][D][binary_sensor:036]: ' PIR': Sending state OFF
[08:38:47][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:47][W][api.connection:071]: ::FFFF:192.168.1.103: Socket operation failed: BAD_INDICATOR errno=11
[08:38:47][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:47][W][api.connection:071]: ::FFFF:192.168.1.103: Socket operation failed: BAD_INDICATOR errno=11
[08:38:50][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:50][W][api.connection:083]: aioesphomeapi (::FFFF:192.168.1.103): Connection closed
[08:38:50][D][api.connection:159]: Home Assistant 2023.2.5 (::FFFF:192.168.1.103) requested disconnected
[08:38:50][D][api:102]: Accepted ::FFFF:192.168.1.103
[08:38:50][D][api.connection:918]: Home Assistant 2023.2.5 (::FFFF:192.168.1.103): Connected successfully
[08:38:52][D][shtcx:100]: Got temperature=26.04°C humidity=41.34%
[08:38:52][D][sensor:127]: ' Temperature': Sending state 23.04301 °C with 1 decimals of accuracy
[08:38:52][D][sensor:127]: ' Humidity': Sending state 46.33759 % with 1 decimals of accuracy
[08:38:56][D][bh1750.sensor:159]: ' Illuminance': Got illuminance=95.9lx
[08:38:56][D][sensor:127]: ' Illuminance': Sending state 95.87106 lx with 1 decimals of accuracy
[08:39:04][D][binary_sensor:036]: ' PIR': Sending state ON

raphael303, 2023-03-11:
It works, entities are there and do as they should. 
But it throws the errors as above in the log. 

And the yaml is also "empty". I guess I'll have to copy paste the yaml from github in there? Or is it supposed to already be there?

EverythingSmartHome, 2023-03-11:
> Hi
>
> I have successfully added the sensor to my HA, but I want to edit the code now. Sure I have ESP-home installed and built a couple of my own devices with ESPs. However I am unsure, how I proceed [...]

The docs for that section are still work in progress unfortunately - feel free to contribute if you wish, would be very welcome.

The code will show up in ESPHome as you already discovered, however it will only be a small section of it - this is the standard ESPHome way and is expected behaviour. The shortened code is all you really need for most tweaks.

If you want the full code, then yes copy and paste from the GitHub and create your own ESPHome firmware instead.

The Home Assistant disconnecting and re-connecting is a bug, update Home Assistant to 2023.3 which should hopefully resolve - PR for that is here: https://github.com/home-assistant/core/pull/89210

closed 2024-07-19 by EverythingSmartHome