EverythingSmartHome/everything-presence-one #30: Two mmw sensors with the free GPIOs - Possible?

raphael303, 2023-03-11:
Hi

The angles of the mmw sensors (40°/100°) are very narrow and for my use case this does really bad. So I want to add another mmw sensor. Is there any reason to assume I couldn't add another mmw sensor with the free GPIOs 18/19/22/23?

Has anybody done it? I'm not exactly fluent with ESP32s, meaning I can't tell if I can have another UART on that GPIOs, but apart from that I'd assume with proper copying and renaming of the mmw code elements in the yaml, this should totally work.

Please let me know, if there is any work already done on this, or stop me if the endeavor is futile.

BradleyFord, 2023-03-11:
Theoretically yes it should be possible.

In am using the spare GPIOs for an extra UART and I2C

EverythingSmartHome, 2023-03-11:
No reason you can't! You should only need 3 of the 4 for this.

closed 2023-03-27 by EverythingSmartHome