EverythingSmartHome/everything-presence-one #35: mmwave false positive - sensitivity 1

neildsb, 2023-03-17:
I am receiving many false positives on the mmWave sensor, is it faulty? I have tuned it down to sensitivity 1, the distance is 3m, off 30 sec, location in a bathroom without windows and a large mirror.....

[image]

Everything_Presence_One
by Everything_Smart_Technology
Firmware: 2023.3.0 (Mar 16 2023, 11:06:07)
Hardware: 1.1.3

additional YAML in ESPHOME from stock

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
  
sensor:
  - platform: wifi_signal
    name: "${room} Wi-Fi"
    update_interval: 600s

bajarrr, 2023-03-17:
Have you tried setting the mmWave On latency to  something more than 0? I had similiar issues to what you describe before I increased that - I think 1 or 2 seconds did the trick for me.

The default for this parameter from the manufacturer of the mmWave sensor is 2.5 seconds.

neildsb, 2023-03-17:
thanks, testing now, have turn the sensitivity back up to 7 while testing

neildsb, 2023-03-17:
That sorted the problem, no false positives for ~30mins, maybe this needs to be a sticky FAQ @EverythingSmartHome EverythingSmartHome, I just noticed the everything-presence-one.yaml default is 1s, I must have changed it on the slider GUI at some point without understanding the consequences :-)

closed 2023-03-17 by neildsb