EverythingSmartHome/everything-presence-one #36: Does mmWave sensor "detect" through glass (windows) ?

reefland, 2023-03-19:
My EP1  is about 16 feet (~4.8m) from a window, but it is looking directly at it.  The mmWave detector is constantly flapping on and off:

[image]

I'm wondering if it can see movement of the trees right outside the window??  I'm home alone, and fairly confident my house is not haunted. However, there seems to be a dance club in my living room when I'm not looking.

* mmWave off-latency: 15 seconds
* mmWave sensitivity: 6
* mmWave distance: 375cm (~12 feet)

These values seem to make sense.  My thought is to reduce sensitivity even more.  (Wow, these sensors are fantastic!)

Is there a way to get information on where it senses movement? (direction / distance / etc)

SDMF-Reaps, 2023-03-19:
I have noticed that mine does not see through windows.  I have a French door that when open blocks the mmwave.  I have another in the dining room where there is a rain stick with feathers attached and these feather set it off.  It took me a long time to figure that out because the motion of the feathers was almost imperceptible.  I'm wondering if you have curtains or something similar that could be moving slightly?

Have you tried turning the sensitivity down? How far did you go?

reefland, 2023-03-19:
I reduced sensitivity from 7 to 6 and the mmWave has not been triggered for 40 minutes.  The windows to the far left side of the sensor do have blinds and may have slight movement from wind gusts (drafts) and forced air room heating.  It is very slight, but is there if you watch for it.

If the sensor can not see beyond the glass, then these are likely the source of movement. These windows are several meters beyond the mmWave distance setting.  I assumed they were beyond EP1 field of view.

aliaghil, 2023-03-27:
I am also facing the same issue; I tried to bring down the sensitivity even to 3 with a distance of 2 meters but failed.
The sensor is next to the TV. However, when we sleep at night, the TV is off, and no one is in the living room, but we still detect movement.
Any help would be appreciated.

[image]

[image]

EverythingSmartHome, 2023-03-27:
Hello!

I added a new page to the user guide with a lot of information about tuning for the space it is in which should help: https://everythingsmarthome.github.io/everything-presence-one/tuning.html

I would suggest you add 0.5s to the on latency setting and see if that helps to clear things up for you.

aliaghil, 2023-03-29:
> Hello!
>
> I added a new page to the user guide with a lot of information about tuning for the space it is in which should help: [...]

This solution solved my problem. As you suggested, I have changed the mmWave On Latency to 0.5.
Thanks a lot for your help.

closed 2024-07-19 by EverythingSmartHome