EverythingSmartHome/everything-presence-one #40: Light intensity bh1750 sensor fails to communicate

gregorycullen1, 2023-03-30:
Hello!

Im having an issue with the bh1750 sensor. No data is being reported to Home Assistant and when looking at the logs, im getting the following error on boot.  

`[E][bh1750.sensor:121]: Communication with BH1750 failed!`

This has happened with both 2023.3.1 and previous firmware versions.

EverythingSmartHome, 2023-03-30:
Hello!

Can you post the startup logs of the device?

gregorycullen1, 2023-03-30:
Yeah for sure. 

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:258]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:175]: Performing I2C bus recovery
[C][uart.arduino_esp32:077]: Setting up UART...
[    90][E][HardwareSerial.cpp:525] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.

[C][status_led:056]: Setting up Status LED...
[D][binary_sensor:034]: ' mmWave': Sending initial state OFF
[D][template.switch:049]:   Restored state ON
[D][switch:013]: 'mmWave Sensor' Turning ON.
[D][switch:056]: 'mmWave Sensor': Sending state ON
[D][template.switch:049]:   Restored state OFF
[D][switch:017]: 'mmWave LED' Turning OFF.
[D][switch:017]: 'mmWave Sensor' Turning OFF.
[D][switch:056]: 'mmWave Sensor': Sending state OFF
[D][switch:056]: 'mmWave LED': Sending state OFF
[D][template.switch:049]:   Restored state OFF
[D][switch:017]: 'uart_presence_output' Turning OFF.
[D][switch:017]: 'mmWave Sensor' Turning OFF.
[D][switch:056]: 'uart_presence_output': Sending state OFF
[D][template.switch:049]:   Restored state OFF
[D][switch:017]: 'uart_target_output' Turning OFF.
[D][switch:017]: 'mmWave Sensor' Turning OFF.
[D][switch:056]: 'uart_target_output': Sending state OFF
[D][number:012]: 'mmWave Distance': Sending state 315.000000
[D][number:012]: 'mmWave Off Latency': Sending state 15.000000
[D][number:012]: 'mmWave On Latency': Sending state 0.000000
[D][number:012]: 'mmWave Sensitivity': Sending state 7.000000
[C][light:036]: Setting up light 'ESP32 Status LED'...
[D][status_led:039]: 'ESP32 Status LED': Setting initital state
[D][status_led:051]: 'ESP32 Status LED': Setting state OFF
[D][light:035]: 'ESP32 Status LED' Setting:
[D][light:040]:   Color mode: 
[C][shtcx:028]: Setting up SHTCx...
[C][shtcx:056]:   Device identified: SHTC3 (0887)
[C][bh1750.sensor:041]: Setting up BH1750 ' Illuminance'...
[E][component:113]: Component bh1750.sensor was marked as failed.
[C][esp32_ble:026]: Setting up BLE...
[D][esp32_ble:042]: BLE setup complete
[D][esp32_ble_server:034]: Setting up BLE Server...
[D][esp32.preferences:113]: Saving 5 preferences to flash...
[D][esp32.preferences:142]: Saving 5 preferences to flash: 4 cached, 1 written, 0 failed
[D][uart_debug:158]: >>> "sensorStartsensorStopsensorStopsensorStop"
[D][uart_debug:158]: <<< "sensorStart\r\n"
[D][uart_debug:158]: <<< "Done\r\n"
[D][shtcx:100]: Got temperature=28.14°C humidity=34.60%
[D][sensor:127]: ' Temperature': Sending state 25.13919 °C with 1 decimals of accuracy
[D][sensor:127]: ' Humidity': Sending state 39.59625 % with 1 decimals of accuracy
[D][esp32_ble_server:071]: BLE server setup successfully
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: C8:F0:9E:EF:DB:70
[D][wifi:048]: Loaded saved wifi settings: Transporter
[D][wifi:386]: Starting scan...
[D][esp32_improv.component:201]: Setting Improv to start
[D][uart_debug:158]: <<< "leapMMW:/>"
[... 194 lines cut ...]
[C][esp32_ble_server:161]: ESP32 BLE Server:
[C][esp32_improv.component:215]: ESP32 Improv:
[C][esp32_improv.component:217]:   Status Indicator: 'NO'
[C][captive_portal:088]: Captive Portal:
[C][mdns:108]: mDNS:
[C][mdns:109]:   Hostname: everything-presence-one-efdb70
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: everything-presence-one.local:3232
[C][api:138]: API Server:
[C][api:139]:   Address: everything-presence-one.local:6053
[C][api:143]:   Using noise encryption: NO
[D][binary_sensor:036]: ' mmWave': Sending state ON
[D][binary_sensor:036]: ' Occupancy': Sending state ON
[C][improv_serial:032]: Improv Serial:
[D][binary_sensor:034]: ' PIR': Sending initial state OFF
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "..\\User_code\\main.c  line:232\r\n"
[D][uart_debug:158]: <<< "reset system\r\n"
[D][binary_sensor:036]: ' mmWave': Sending state OFF
[D][uart_debug:158]: <<< "             \r\n"
[D][uart_debug:158]: <<< "leapMMW:/>"
[D][shtcx:100]: Got temperature=28.16°C humidity=34.61%
[D][sensor:127]: ' Temperature': Sending state 25.15788 °C with 1 decimals of accuracy
[D][sensor:127]: ' Humidity': Sending state 39.60541 % with 1 decimals of accuracy
[D][binary_sensor:036]: ' Occupancy': Sending state OFF`

EverythingSmartHome, 2023-03-30:
Thanks. It looks like the BH1750 isn't even being detected through the i2c scan.

Seems like a hardware problem so please reach out to support with your order number and we can arrange a replacement. Use the contact page on the shop and we will get back to you ASAP

closed 2023-03-30 by gregorycullen1