EverythingSmartHome/everything-presence-one #41: Could not update firmware: Failed to open serial port

Quentame, 2023-03-31:
Hi !

I got an EP1 (Everything Presence One) from [Everything Smart Home](https://www.youtube.com/@EverythingSmartHome) ([Hardware overview](https://everythingsmarthome.github.io/everything-presence-one/hardware-overview.html)), and trying to update via the official tool, using ESP Web Tools, [here](https://everythingsmarthome.github.io/everything-presence-one/updating.html), but it's not working.

**How to reproduce**
1. Go to the firmware installation page: https://everythingsmarthome.github.io/everything-presence-one/updating.html
2. Click connect
3. Chrome is displaying this dialog: 
[image]
4. Whenever I chose the first or the second option, I got the same Javascript error, with no console logging: `Failed to execute 'open' on 'SerialPort': Failed to open serial port.` 
[image]

I also tried on official ESP tools here:
- https://esphome.github.io/esp-web-tools/
- https://web.esphome.io/

==> With the same results.

**Environnement**
- Windows 10 Entreprise 21H2
- Chrome 111
- The cable is passing data + power, verified by connecting an Android phone to my machine.

**Additional informations**
I think the COM3 line while selectiong serial ports was not there at first. Then, Windows installed some kind of driver when I plugged in the EP1, then the line appear.
When having this troubleshouting dialog, I installed CP2102 drivers, but it did not change, even after a reboot. I don't know what kind of chip it is, and I think it would be written in the EP1 doc if I needed to install a driver.
[image]

Thanks

Quentame, 2023-03-31:
I first created the issue here: https://github.com/esphome/esp-web-tools/issues/353

But balloob said "This is a driver issue. Please contact the manufacturer about what drivers you need to install.".

So here I am 😄

Quentame, 2023-03-31:
Haven't seen somewhere talking about driver to install in the doc, 2 issues here, but not the same thing (Linux rights + Mac ARM arch).

Or maybe it's related to the organisation managing the laptop.

Quentame, 2023-03-31:
Will try on Mac Intel + Windows 11 (non Entreprise)

EverythingSmartHome, 2023-03-31:
CP2102 is for a silicon labs USB to Serial adapter which isn't used in the EP1 so installing drivers for that won't work. The chip is a CH340, but the drivers for this are generally included with Windows 11 so no need to install them (usually, unless your work uses special driver packs). The link for the CH340 is in the screenshot you have above.

Quentame, 2023-03-31:
It worked, thanks !

I added a warning into the install page to avoid someone else reproducing the error, PR #42

PR #42 (2023-04-28): Add warning when USB serial is not detected because of a missing driver
Code not tested, pretty much confident except for the tick ` inline code display.

Fixes #41
closed 2023-04-28 by EverythingSmartHome