EverythingSmartHome/everything-presence-one #47: ESPresense

chrispi68, 2023-04-01:
Is it possible to install ESPresence as firmware? Which setting do I have to use to get the radar and the IR working?

I am a little disappointed that MQTT is not supported and I do not use Home Assistant. ESPresence has MQTT built in.

I also don't want to recompile the firmware of ESPHome just to enable MQTT.

EverythingSmartHome, 2023-04-01:
MQTT is supported, just create your own ESPHome config.

I don't use ESPresence but last time I checked it didn't support all the necessary options and settings, it might work but you will probably be limited. But again, I haven't checked for a while.

chrispi68, 2023-04-01:
> MQTT is supported, just create your own ESPHome config.

How can I change the ESPHome config? The device is logged into the WLAN. But I can't reach any web interface and I can't access it via SSH. I know the IP address.

EverythingSmartHome, 2023-04-01:
You need to create a new config and upload that to the device. You can follow the instructions on the ESPHome website: 

https://esphome.io/guides/getting_started_command_line.html

Or there is videos of getting started

chrispi68, 2023-04-01:
Sorry, this is too complicated for me. I am stuck at the first question "Are you using an ESP32 or ESP8266 platform?".

Is there a chance to get some information on which GPIOs the PIR and Radar are connected so I can set up ESPresense?

EverythingSmartHome, 2023-04-03:
Pinout is as follows:

SDA: GPIO26
SCL: GPIO25
ESP32 LED: GPIO32
mmWave Output: GPIO15
PIR: GPIO33

DanielWe2, 2023-04-08:
> Sorry, this is too complicated for me. I am stuck at the first question "Are you using an ESP32 or ESP8266 platform?".
> 
> Is there a chance to get some information on which GPIOs the PIR and Radar are connected so I can set up ESPresense?

the esphome yaml files used are in this repository. I think you can leave most of it as it and just add mqtt

closed 2023-07-27 by EverythingSmartHome