EverythingSmartHome/everything-presence-one #48: lux sensor - accuracy

reefland, 2023-04-03:
Trying to understand what the lux sensor is telling me. Your documentation states "Illuminance is measured from the on-board light sensor in lux.".

Looking at <https://en.wikipedia.org/wiki/Lux> they give some examples of lux values:
[image]

I'm showing `0.9 lx` in my front entrance, seems way brighter than a full moon at night.

Here is your example in the docs:
[image]

Were you sitting in the dark with that device?

My living room, shows `12.0 lx` in a room full of windows, shades open, bright sunny spring day here.  According to the Wiki, a `20` starts `Public areas with dark surrounding`.  I'm not even close to that value.

Scratching my head on what to do with this.  Seems like a multiplier is off/needed maybe?  The values are so low I can't make decent automatons based on these values.  

I'm using the provided 3D printed cases, openings are not blocked.

EverythingSmartHome, 2023-04-03:
Hello!

Lux can be a bit of a funny thing to measure really, for example you can take a lux reading at one side of a room and get completely different number to the other side of the room - it's very hard to compare numbers. What matters about Lux readings is that they are repeatable, i.e it doesn't read 100 lux one day, then the next day in the exact same position and conditions it reads 5.

That said, 12 does seem pretty low, but it depends on where you have it mounted/positioned. Do you have in a corner with the light sensor facing into the wall/corner?

Do you have a picture you could share?

As a test, you can also move the sensor into a different position, rotate it a little and/or remove the cover and see what the readings are.

reefland, 2023-04-08:
It is mounted just off center to one of the walls. About 7 foot high, over looking a couch area it is monitoring for presence detection.  Two windows to the left side. one is perpendicular to field of view (about 10 foot away), other is about 35/45 degree about 14 foot distance, and one is directly in front of it (~16 feet away, I asked about device seeing movement through glass in #36 with this window).

I understand your point about its more of a trend then a direct measurement.  I can work with that. I'm sure it will vary more as season change, and Spring kicks in and trees get leaves. 

[image]

I have the correlation between sun/cloud and LX spikes.  Values are just not as high as I was expecting.

closed 2023-04-08 by reefland