EverythingSmartHome/everything-presence-one #50: 2 Devices stuck in boot loop after Installing firmware for Home Assistant

Techtony96, 2023-04-06:
Received my 2 everything presence ones today, and ran the Home assistant installer. When trying to connect to wifi, a list of wifi networks is not populated, and when manually entering in the SSID/PW, it connects forever. When looking at the logs, it has the following error and then reboots, and gets the same error again.

Full error: [error.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/11165201/error.txt)

```
[C][improv_serial:032]: Improv Serial:

abort() was called at PC 0x401ed4ef on core 1

Backtrace:0x40083c19:0x3ffebbf00x4009364d:0x3ffebc10 0x40098d89:0x3ffebc30 0x401ed4ef:0x3ffebcb0 0x401ed536:0x3ffebcd0 0x401ee5af:0x3ffebcf0 0x401ecf36:0x3ffebd10 0x401ecf05:0x3ffebd30 0x400f04de:0x3ffebd50 0x400ef7ea:0x3ffebd70 0x400efae0:0x3ffebec0 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```

djgalati, 2023-05-06:
im having the same issue, ive attached full logs below of what the output is. For reference, my device had the smartthings SW installed and i decided to switch it to the home asisstant version and this started happening
[esp-web-tools-logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/11411528/esp-web-tools-logs.txt)

EverythingSmartHome, 2023-05-06:
> Received my 2 everything presence ones today, and ran the Home assistant installer. When trying to connect to wifi, a list of wifi networks is not populated, and when manually entering in the [...]

Hi, my apologies I completely missed this, did you get it sorted?

EverythingSmartHome, 2023-05-06:
@djgalati so it was connecting fine previously then you switched to the HA firmware and now you are having issues?

Can I check your installing from the user guide or your own ESPHome dashboard?

Sphazz, 2023-12-16:
I'm having the same issue with my new Everything Presence One. I received my EPO last week and while attempting to install the firmware via the user guide I cannot select a wireless network. I checked my logs and it's almost exactly the same as what others have posted here with the device boot looping after flashing the firmware. I tried a few different cables in case that was the issue, as well different USB 2.0 and USB 3.0 but same problem.

Here are some snippets where _I_ see some issues, this is after the first reboot.

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:326]: Log initialized
[C][ota:473]: There have been 3 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
```

```
[D][switch:016]: 'mmWave sensor' Turning OFF.
[D][uart_debug:158]: >>> "sensorStopsensorStopsensorStopsensorStop"
[D][uart_debug:158]: <<< "sensorStop\r\n"
[D][uart_debug:158]: <<< "sensor stopped already\r\n"
[D][uart_debug:158]: <<< "Error\r\n"
[  3160][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[E][shtcx:094]: sensor read failed
[D][sensor:094]: 'Temperature': Sending state nan °C with 1 decimals of accuracy
[D][sensor:094]: 'Humidity': Sending state nan % with 1 decimals of accuracy
[W][component:214]: Component shtcx.sensor took a long time for an operation (1.02 s).
[W][component:215]: Components should block for at most 20-30ms.
```

```
[I][app:102]: ESPHome version 2023.10.6 compiled on Nov  9 2023, 10:09:01
[I][app:104]: Project Everything_Smart_Technology.Everything_Presence_One version 1.1.7
```

```
C][wifi:546]: WiFi:
[C][wifi:382]:   Local MAC: A8:42:E3:1F:FE:A8
[C][wifi:383]:   SSID: ''[redacted]
[C][wifi:384]:   IP Address: 0.0.0.0
[C][wifi:386]:   BSSID: 00:00:00:00:00:00[redacted]
[C][wifi:387]:   Hostname: 'everything-presence-one-1ffea8'
[C][wifi:389]:   Signal strength: 0 dB ▂▄▆█
[C][wifi:393]:   Channel: 8
[C][wifi:394]:   Subnet: 0.0.0.0
[C][wifi:395]:   Gateway: 0.0.0.0
[C][wifi:396]:   DNS1: 0.0.0.0
[C][wifi:397]:   DNS2: 0.0.0.0
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: everything-presence-one-1ffea8
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: everything-presence-one.local:3232
[W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[C][api:139]: API Server:
[C][api:140]:   Address: everything-presence-one.local:6053
[C][api:144]:   Using noise encryption: NO
[C][improv_serial:032]: Improv Serial:

abort() was called at PC 0x401ee633 on core 1

Backtrace:0x40083c19:0x3ffeb8900x4009364d:0x3ffeb8b0 0x40098d89:0x3ffeb8d0 0x401ee633:0x3ffeb950 0x401ee67a:0x3ffeb970 0x401ee773:0x3ffeb990 0x401ee6d2:0x3ffeb9b0 0x401ee7dd:0x3ffeb9d0 0x400f1462:0x3ffeb9f0 0x400f076e:0x3ffeba10 0x400f0a64:0x3ffebb60 

ELF file SHA256: 0000000000000000
```

Then it reboots again until it's tried rebooting too many times.

I do have something to add though, I was able to get my EPO working by flashing the Beta version of the EPO firmware. Now, it's a bit buggy and I can't tell if it's the _beta_ part of the firmware or the device itself. But false detections is a problem, but not with the mmWave sensor which I've been able to dial in, but the PIR which keeps triggering in the middle of the day in an empty room with the door closed. I haven't been able to rule out interference, because it could be environmental, but I don't know if it's the _beta_ software or something related to being unable to boot on the non-beta firmware.

Hopefully this information is helpful to someone.

vzander925, 2023-12-17:
Hello,

Mine was a power issue. I used a different PC and it came up just fine. Still working on installing it as it was a work computer I will need to borrow, but I can keep you posted if anything changes or I have another update

Sphazz, 2023-12-26:
> Hello, Mine was a power issue. I used a different PC and it came up just fine. Still working on installing it as it was a work computer I will need to borrow, but I can keep you posted if anything changes or I have another update

I appreciate the information. It's the holiday season so I've been busy, but today I tried your recommendation on three separate PCs and a variety of ports (including USB C 3.1 with power delivery) and three different cables including with the same results of boot looping on the normal firmware, but flashing fine on the beta. So we might've had different issues with the same symptoms or all of my computers just can't power the EP1 properly.

I've given up at this point since the beta firmware seems to work and only has some occasional false positives on the PIR but overall is relatively stable.

Some additional information, I'm running _Firmware: 2023.10.6 (Nov 9 2023, 10:10:00)_ with _Hardware: 1.3.0b_. If anyone else encounters this issue, try a different computer as recommended by vzander925 or if that doesn't work try the Beta firmware 2023.10.6.

I still love this device.

djgalati, 2024-01-07:
> @djgalati so it was connecting fine previously then you switched to the HA firmware and now you are having issues?
> 
> Can I check your installing from the user guide or your own ESPHome dashboard?

sorry I hadn't responded earlier. I believe my USB cable had a bad connection and I ended up swapping it out and that resolved my issue.

Sphazz, 2024-01-10:
@EverythingSmartHome 
Today my EP1 decided to boot loop and hasn't been able to recover for any extended period of time. There was nothing to trigger it, just seems to have happened out of the blue, no update or changes to configuration and while there was no presence to be detected or anyone was even home.

I was running 2023.10.6 (Nov 9 2023, 10:10:00) beta firmware with hardware 1.3.0b. Please see my other comments here.

Previous to this, for both flashing the firmware and powering my EP1, I have tried more USB C cables than I can count, including high quality cables that I've used for 45w charging, data cables that I've used to successfully transfer data between phones, three different computers with USB 2 and 3 ports to flash the non-beta firmware and it always fails. I was previously able to successfully flash the beta firmware, but now that firmware is also causing my EP1 to boot loop. I am now currently running the beta firmware 2023.12.5 (Jan 9 2024, 00:32:12) and am still having issues with boot looping. I did try the non-beta firmware and that immediately boot looped.

I was always concerned that I couldn't run the non-beta firmware, but the beta worked. I can't find any point of failure that would cause my EP1 to suddenly boot loop on the beta firmware as it was on the non-beta.

Here is the device logs of it boot looping on v2023.10.6:
```[D][sensor:094]: 'Target 1 Distance': Sending state 0.40600 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 1 SNR': Sending state 56.65600  with 1 decimals of accuracy
[D][sensor:094]: 'Target 2 Distance': Sending state 1.17200 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 2 SNR': Sending state 2.96900  with 1 decimals of accuracy
[D][sensor:094]: 'Target 3 Distance': Sending state 1.64800 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 3 SNR': Sending state 2.65600  with 1 decimals of accuracy
[D][sensor:094]: 'Target 4 Distance': Sending state 2.76600 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 4 SNR': Sending state 2.62500  with 1 decimals of accuracy
[W][component:214]: Component uart took a long time for an operation (0.06 s).
[W][component:215]: Components should block for at most 20-30ms.
[D][uart_debug:158]: <<< "$JYRPO,4,1,0.398, ,52.875, , *\r\n"
[D][UART:287]: Target Count: 4, Target Index: 0
[D][uart_debug:158]: <<< "$JYRPO,4,2,1.602, ,2.063, , *\r\n"
[D][UART:287]: Target Count: 4, Target Index: 1
[D][uart_debug:158]: <<< "$JYRPO,4,3,2.836, ,2.219, , *\r\n"
[D][UART:287]: Target Count: 4, Target Index: 2
[D][uart_debug:158]: <<< "$JYRPO,4,4,3.148, ,1.719, , *\r\n"
[D][UART:287]: Target Count: 4, Target Index: 3
[D][sensor:094]: 'Target 1 Distance': Sending state 0.39800 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 1 SNR': Sending state 52.87500  with 1 decimals of accuracy
[D][sensor:094]: 'Target 2 Distance': Sending state 1.60200 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 2 SNR': Sending state 2.06300  with 1 decimals of accuracy
[D][sensor:094]: 'Target 3 Distance': Sending state 2.83600 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 3 SNR': Sending state 2.21900  with 1 decimals of accuracy
[D][sensor:094]: 'Target 4 Distance': Sending state 3.14800 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 4 SNR': Sending state 1.71900  with 1 decimals of accuracy
[W][component:214]: Component uart took a long time for an operation (0.06 s).
[W][component:215]: Components should block for at most 20-30ms.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
I (29) boot: ESP-IDF 4.4.5 2nd stage bootloader
I (29) boot: compile time 10:11:59
I (29) boot: chip revision: v3.1
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (70) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (77) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (85) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (92) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (100) boot: End of partition table
```

Now on v2023.12.5
```
[D][sensor:094]: 'Target 1 Distance': Sending state 0.30500 m with 2 decimals of accuracy
[D][sensor:094]: 'Target 1 SNR': Sending state 4.78100  with 1 decimals of accuracy
[... 12 lines cut ...]
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
I (29) boot: ESP-IDF 4.4.5 2nd stage bootloader
I (29) boot: compile time 00:33:13
I (29) boot: chip revision: v3.1
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
[... cut ...]
I (77) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (85) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (92) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
```

My EP1 is completely inoperable at the moment, it'll boot loop for awhile, then work for a short duration, then boot loop again for an extended period of time. Any guidance would be appreciated, if this is not the correct avenue for support, please direct me to the correct location.

EverythingSmartHome, 2024-07-19:
Please open a new issue if this is still a problem as there has been a lot of updates since

closed 2024-07-19 by EverythingSmartHome