EverythingSmartHome/everything-presence-one #51: PoE Ethernet Support

Rogue136198, 2023-04-08:
I've been messing around with esp32 boards lately and I came across the ESP32-POE-ISO, an ESP32 that can be powered by PoE. This has me wondering. Would it be possible to build a EP1 using this board? I like the concept of the EP1 but I hate adding more wifi devices to my network. Also if I'm going to run a cable to power the EP1 it may as well also handle data... What are the thoughts on being able to make a PoE EP1? I may take a stab at DIYing.

techydude, 2023-04-08:
that would be one buttugly hack job to desolder the as-designed-in ESP32 module, and hack in that ESP32-POE-ISO.  you'd probably have to mount that ESP32-POE-ISO on the back-side of the EP1, and have a bunch of wires linking them.

you'd want to do a thorough comparison of ALL the GPIOs that the EP1 uses from its ESP32 module (which means tedious drawing out a scematic to see what's connected to each of the ESP32 module's GPIOs), and those available on that ESP32-POE-ISO, before touching anything; i suspect you'd run into problems pretty quick, i remember an ESP32 Ethernet project on CrowdSupply about 4 years ago (supposedly the first open-source ESP32 board with Ethernet), something about the ESP32's GPIO config to support Ethernet was a bit weird.  and me not being all that familiar with ESPHome, I don't even know if it supports Ethernet!?

Rogue136198, 2023-04-08:
> that would be one buttugly hack job to desolder the as-designed-in ESP32 module, and hack in that ESP32-POE-ISO. you'd probably have to mount that ESP32-POE-ISO on the back-side of the EP1, and have a bunch of wires linking them.

I'd have no intention to do it this way. If I were to do this I'd get the components individually and mount them directly to the ESP32-POE-ISO or more likely a daughter board. 

> not being all that familiar with ESPHome, I don't even know if it supports Ethernet!?

It does... https://esphome.io/components/ethernet.html

Ideally I'd be able to get the required POE components to connect Ethernet and power to the EP1 with minimal modifications.

techydude, 2023-04-08:
nice to see someone doing their hacks properly :-)

Rogue136198, 2023-04-08:
> nice to see someone doing their hacks properly :-)

Yep, just ordered the components, now to find my breadboard....

techydude, 2023-04-12:
Perhaps the "EP2" can offer a PoE version instead of USB-C :-), hey @EverythingSmartHome ?!

EverythingSmartHome, 2023-04-12:
It's already well underway and close to release :)

It's not a replacement for USB-C, that wouldn't be great as not everyone has PoE, but rather an alternative

Rogue136198, 2023-04-12:
> It's already well underway and close to release :)
> 
> It's not a replacement for USB-C, that wouldn't be great as not everyone has PoE, but rather an alternative

Interesting, very interesting. Well I may need to hold up on mine a bit. I am very interested to see what you have in the works.

amatsoukas, 2023-06-17:
> It's already well underway and close to release :)
> 
> It's not a replacement for USB-C, that wouldn't be great as not everyone has PoE, but rather an alternative

What would be also nice, is to have a voltage regulator on the 5V pins, that can withstand a wider range of voltage (5-12V). In my case, i used the existing wiring in my house for my old alarm system which i do not use anymore, to power the ep1 with 12v (i could not use 5v power, since there would be considerable voltage drop in the wiring until it reaches the ep1), using an external voltage regulator connected just before the ep1 to provide 5v to the board. It works, but it is ugly :P. Would be much better to be able to directly connect the wiring to the board.

EverythingSmartHome, 2023-06-17:
Also already a thing :D

TheBlueKingLP, 2023-06-20:
> It's already well underway and close to release :)
> 
> It's not a replacement for USB-C, that wouldn't be great as not everyone has PoE, but rather an alternative

@EverythingSmartHome How close is the release? 1 Month? 1-3 months? within a year?

I just saw your restock post on YouTube and decided to buy it but if it is releasing soon then I would wait for the PoE version.

Any ETA would be greatly appreciated.

taw123, 2023-12-25:
Back in [April](https://github.com/EverythingSmartHome/everything-presence-one/issues/51#issuecomment-1505277388) you mentioned you were on the cusp of releasing a POE based EP 1 (aka EP2?)...  

I too want to power all my sensor devices over PoE and where possible appreciate the option to move as much traffic possible to a hardwire (not to mention being the ONLY viable option if someone wants to employ in a security application).  

Any update on your dev schedule? I have 3 FP2's that I am now up to deploying over in the next 2 months and considering either 8-10 more (FP2 or comparable) with my next wiring/network project.

Thanks for ANY update on this! Really want to support this (adding code and funding/purchase).
--T

EverythingSmartHome, 2023-12-26:
@taw123 Yes actually, I just released some information about it on Twitter/YouTube last week: https://www.youtube.com/post/Ugkxg7zPI5tcndLtAiae3jCv9aYbKzONA8BS

Not sure how long now but this is pretty close to final

kisst, 2024-04-24:
Just read this post and then the YT post , and very much looking forward for the release , I already use PoE splitters to power my EP1 so direct native support and no WiFi will make a difference . 
One thing I think was a bit of a hit and miss on sensor side is UV , while brightness does scale most of the time with UV , and combined with external weather sensors and or with light states it is enough for most use-case I think it would been a great addition to the pile of sensors.
Otherwise just eagerly looking forward for the release .

taw123, 2024-09-07:
Still upgraded one of my switches in anticipation, so still hoping :)

tantalic, 2024-12-11:
@EverythingSmartHome any updates on the Everything Presence Pro? Really looking forward to getting a few of these installed. Thanks!

SamueldaCostaAraujoNunes, 2025-06-16:
I would love to know if there is any update on this topic! It would really be a great blessing to include sensors in the network directly, without filling up the wifi and still being able to include them in my UPS lol

ELLIOTTCABLE, 2025-11-28:
Oh man, really hope this hasn't been canned. Gonna hold off on all the excellent Black Friday deals, because I sorely want PoE, and a PIR *with* the multi-zone mmWave instead of having to choose between them …