EverythingSmartHome/everything-presence-one #52: proper way to power from 5v headers

travi, 2023-04-13:
i havent been able to successfully power up the sensor using direct 5v. 

I'm looking to do a fairly permanent installation where usb-c wont be convenient, so i'm hoping to use the headers instead. i switched the jumper based on my understanding from https://shop.everythingsmart.io/en-us/products/everything-presence-one-kit, but didn't find more specific details in https://everythingsmarthome.github.io/everything-presence-one/ to confirm that i was approaching it correctly. 

i'm using a 5v, 1500 mA power supply with a barrel adapter to connect the dupont wires to the headers. i do see the led on the mmWave board come on briefly, but it doesn't see to boot fully. if i switch the jumper back and connect to usb-c power again, it boots fully without trouble.

is there something that i am unknowingly just doing wrong? how can i investigate further?

ColColonCleaner, 2023-04-19:
@travi I'm powering all of my EP1s using a 12v lighting transformer, running the 12v through standard lamp cord and terminating with a 9-24 USB adapter. Only the white USB sticks out of my ceiling to reach the device. 12v lighting cord can be spliced easily and splicing adapters are readily available, so you can run a single mainline and branch off of that to each of your rooms.

[image]

ColColonCleaner, 2023-04-19:
To be clear, the lighting transformer is in my basement, hooked up to the same central UPS that powers my network so it won't lose power during an outage. The lamp cord runs up through a wall into the attic in the same channel I use for ethernet. Once in the attic it runs across the length of the house with splices out to each EP1 location.

ColColonCleaner, 2023-04-19:
Video of the preliminary set up. https://youtu.be/8rTABDTAcgE

travi, 2023-04-19:
@ColColonCleaner thanks for the info. it is useful and worth considering, but it doesnt answer the primary thing that I'm trying to understand. the 5v input headers are separate from the usb port. i'm trying to understand why I haven't been successful powering with those headers. i get the potential of voltage drop with a longer cable run, but so far, i've been using jumper wires directly from the barrel adapter. higher voltage with a step down could be useful once getting closer to the permanent installation, but i'm still just trying to get it to power on using the headers instead of the usb port.

EverythingSmartHome, 2023-04-20:
Hi! Have you setup the EP1 via USB-C first before attempting to power via the headers?

What you are doing should work fine without issues, that is the way I ran one of my EP1s for a long time. If you want to join the discord and post in the EP1 section, it might be easier to help you troubleshoot.

travi, 2023-04-25:
I did set up with USB-C first and had no trouble getting it integrated into ESPHome and Home Assistant when powered that way. Not having luck after switching the jumper and powering through the headers instead.

Which discord should I join? I'm probably overlooking the link somewhere.

EverythingSmartHome, 2023-04-25:
That's really strange, there isn't much to really go wrong. Have you checked your power supply with a meter to confirm how much voltage you are actually getting? As well as the polarity the correct way around?

Do you have a different supply you could test with?

travi, 2023-05-09:
sorry for the delayed follow up. i was finally able to try things out again and ended up getting it working with a 12v power supply + a step down voltage converter to 5v since i wasn't able to find another 5v power supply in my stash. 

i think that is enough to confirm that the issue ended up being a problem with the power supply that i was trying to use and completely on my end. i think it is safe to close this out. sorry for the noise

closed 2023-05-09 by travi