EverythingSmartHome/everything-presence-one #53: Adding a 5v sensor?

acronce, 2023-04-16:
I'm interested in adding a PMSA003 dust sensor to my Everything Presence One. I realize that I'll have to extend the config to support this (I'm a developer, and experienced with ESPHome, so no problem there), and I'll need a custom case (got a 3D printer, so also no problem).

All I need to drive the AQI sensor is 5 volts and a single IO. But the breakout header only has 3.3v (plus ground and the spare IOs).

I assume that the 5v headers at the bottom of the board are inputs (when the jumper selects v5 direct instead of USB). Without examining the schematic I'm not sure if that header also provides 5v out when powered by USB (as I intend to do).

Perhaps it would be safer for me to tap the 5v on the microwave header that I'm not using (there are two of them)?

Thanks in advance for advice for tapping 5v.

BradleyFord, 2023-04-16:
If you look in the manual the 5v at the bottom is an alternative input.

The other 5v output is at the mmWave, but I know if it is enough to power
the PM2.5 sensor as that could be sure hefty.

One idea for you is rather than powering your board via USB-C, do it with
5v so you can also use that for the extra sensor.

travi, 2023-04-19:
> If you look in the manual the 5v at the bottom is an alternative input.

@BradleyFord would you mind linking to what you mean by "the manual"? as mentioned in #52, i've been attempting to power using those headers, but have been unsuccessful so far. I have not found additional detail to set me straight in either https://everythingsmarthome.github.io/everything-presence-one or https://shop.everythingsmart.io/en-us/products/everything-presence-one-kit

ColColonCleaner, 2023-04-19:
@travi I'm powering all of my EP1s using a 12v lighting transformer, running the 12v through standard lamp cord and terminating with a 9-24 USB adapter. Only the white USB sticks out of my ceiling to reach the device. 12v lighting cord can be spliced easily and splicing adapters are readily available, so you can run a single mainline and branch off of that to each of your rooms.

[image]

closed 2024-07-19 by EverythingSmartHome