EverythingSmartHome/everything-presence-one #54: EP1 is unable to connect to hidden WiFi networks

ColColonCleaner, 2023-04-19:
I am currently only able to get EP1s to connect when the IoT SSID i use is broadcasting publicly. If I change the network to hidden, EP1s cannot connect to it anymore.
Keep in mind the configs are not changing on the device, all I do is toggle the network between hidden and public and the EP1s are only able to connect to the public version.
There are 50 Shelly devices and a dozen other assorted IoT devices on this network so I think this is only a problem on the EP1 side.

What is the first thing to check so I can resolve this?

EverythingSmartHome, 2023-04-20:
There is already a way to do this in the stock config, it's not a problem but rather an ESPHome design.

Add the following line to your EP1 config under the substitutions section:
```
substitutions:
  hidden_ssid: "true"
```

This will allow ESPHome to use fast connect, a feature described [here](https://esphome.io/components/wifi.html#configuration-variables)

ColColonCleaner, 2023-04-20:
Understood, I'll add that option and confirm it's working. Thanks!

closed 2023-04-28 by EverythingSmartHome