EverythingSmartHome/everything-presence-one #55: Raw sensor number from mmWave

ColColonCleaner, 2023-04-19:
Hey @EverythingSmartHome, just got 13 EP1s and I'm trying to fine tune the mmWave thresholds. This process would be a lot easier if I could see the raw motion value from the mmWave sensor to see how the different units compare and are reporting, and set the threshold sensitivity more easily.
Can a sensor for this be exposed?

ColColonCleaner, 2023-04-23:
This is proving to be a very difficult process.

In most of my rooms even when they are completely empty (no people, no pets, no fans, no computers, no breeze, no windows, no lights, and with the door closed) the mmwave sensor just sits on detected status forever.
I have the distance dialed in, the sensor is up on a wall away from vibration, and i have the sensitivity turned all the way down to 0 or 1.
The only thing that makes the sensor clear is turning the distance down to below 100cm. If i increase to 150 it goes right back to detected when there's nothing but open air in front of the sensor. I'm very confused at this point.
Any help would be appreciated.

plaxtika, 2023-04-24:
Yes , me too. The mmWave always detects something - but theres nothing. Change the on latency isn't the right thing because now the mmwave turns on the light too late because he noticed me too late. 

this is really annoying...

EverythingSmartHome, 2023-04-24:
@ColColonCleaner have answered your questions on Discord.

@plaxtika Is it not suitable to allow for 0.5s latency and then use the PIR instead to turn lights on? That's what most users do as the PIR is better suited for turning lights on amyways

closed 2023-07-27 by EverythingSmartHome