EverythingSmartHome/everything-presence-one #57: WiFi signal strength sensor

ColColonCleaner, 2023-04-20:
Would be great for the EP1 to expose a WiFi signal strength (RSSI) sensor to HA. Good for debugging dead zones or interference. Shelly relays do this and it's a really nice feature to have.

DutchDeffy, 2023-04-23:
Hi, 
Because the EP1 is based on ESPhome, you can do this yourself. ([docs](https://esphome.io/components/sensor/wifi_signal.html)).

ColColonCleaner, 2023-04-23:
Thanks! I should have followed up that I found it and am now using a fully custom config rather than just the substitutions. It's great that ESPHome has this available.

Would still be good to keep this issue open so it can be added to the official config. I'll open a PR if I get the chance. Thanks again!

closed 2023-07-27 by EverythingSmartHome