EverythingSmartHome/everything-presence-one #58: OTA Updates

ejbrownie, 2023-04-25:
I have two EP1s that I am trying to update to the latest version of ESPHome. They are currently on version 2023.3.2 and I'm trying to update them over the air to 2023.4.1. When I press the safe mode button, the device reboots but it doesn't appear that anything else changes. The OTA update states the update failed because the client resets the connection. 

Logs are attached.  The logs are from after the Safe Mode button was pressed. What am I missing?

https://docs.github.com/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax
[logs_everything-presence-one-6f1878_logs (1).txt](https://github.com/EverythingSmartHome/everything-presence-one/files/11323253/logs_everything-presence-one-6f1878_logs.1.txt)

EverythingSmartHome, 2023-04-26:
Hello,

When you press safe mode, this will disable some of the core components on the device (including connection to HA) to allow it to complete the update properly, it's only required when updating over wireless (OTA). If you are updating over USB then you do not need to use safe mode.

I'm not sure if this is the process you are already doing but you should press the safe mode button once and then immediately start the update process which should complete without issues.

ejbrownie, 2023-04-26:
That’s what I’ve been trying and it still fails. I am doing all this through the HA UI. Is there a better method to do the update since pressing safe mode disconnects it from HA? I’m obviously missing a step here.

EverythingSmartHome, 2023-04-26:
It disconnecting from Home Assistant is the intended behaviour when you press the safe mode option, it leaves the OTA function intact however.

One issue I've seen on slower machines is the compile time takes too long and so the device reboots out of safe mode before ESPHome finishes compiling. You could try hitting upload and waiting till just before it tried to upload before hitting the safe mode button

ejbrownie, 2023-04-27:
This worked perfectly. I waited until the compile was complete then pressed the Safe Mode button. The OTA update was successful.

EverythingSmartHome, 2023-04-28:
Glad to hear it!

cbho, 2023-09-29:
Hi,
would it be possible to make a setting that increases the time the device is in safe mode?
Until now, I haven't experienced a single time, that compiling etc. doesn't take longer than the time the device is in safe mode if I press it when I start updating.
I'm aware of the solution with sitting and waiting until the compilation is done and new firmware is uploaded - but that's kind of unproductive just sitting there watching...
Also, it is necessary to do this for each device...
Increasing the time the 
Also, this would make it possible to just press update all followed by setting them all in safe mode would eliminate the need for (in my case) waiting and watching three times in a row...

Regards,
Chr.