EverythingSmartHome/everything-presence-one #60: Not clear how to use this for actual presence. This is just a motion sensor.

iloveicedgreentea, 2023-05-01:
This is marketed as presence detection, but will stop detecting me if I am sitting still. This doesn't really work well for light detection without resorting to cheap hacks like assuming time spent in room, hoping for more motion detections, etc. I was under the impression it would continuously detect presence, not just detect motion.

EverythingSmartHome, 2023-05-01:
This is a presence device, it does continuously detect presence. You haven't provided any information about what you have done, where the sensor is place, what settings your using, how you have attempted to tune it, or the environment it's in for starters.

More likely it's a case of not tuning it properly, can you provide some additional info?

iloveicedgreentea, 2023-05-01:
Hmm, okay. I will take that back then. I tried a bunch of configs according
to the tuning guide but maybe it is related to the position or aiming. I
will keep experimenting. It doesn’t have any false detections but the
sensors will stop showing as detected if I stay still, at about 20ft and it
is set to about 24ft of detection. Are there limits on how much of the body
needs to be visible to the sensors?

EverythingSmartHome, 2023-05-01:
What sensitivity are you using? What is the on latency set to? Discord might be best for troubleshooting

iloveicedgreentea, 2023-05-22:
Will do some more troubleshooting on my end. What is the discord?

closed 2023-05-22 by iloveicedgreentea