EverythingSmartHome/everything-presence-one #62: mmWave Sensor detecting with no movement

oneseventhree, 2023-05-11:
Hi,

Just recently purchased this and I am confused why sometimes the mmWave detects motion in an empty room.

I can see my lights trigger and when I check HA it says the mmWave went off.

The motion detector has not found motion even though that detects motion when the mmWave doesn't

This is very intermitttent and I can't put a finger on it.

I am using the mmWave to trigger my lights to turn on because it tthe motion sensor is way too sensitive and from what I can see that can't be adjusted?

Is it bad to use the mmWave to trigger something on?

The mmWave sensitivity is at 0. I can't even trigger the "fault" by dancing around outside of the range of the mmWave but nothing is moving inside the range for it to trigger.

Any thoughts?

EverythingSmartHome, 2023-05-11:
Hello,

Most likely you need to tune the sensor properly for the room, most of the time users don't realise how sensitive mmWave is because they compare it to a regular PIR. Usually people think that something isn't moving in the room when really there is movement and they just didn't realise it.

I've created a tuning guide here: https://everythingsmarthome.github.io/everything-presence-one/tuning.html

The PIR is extremely accurate and I've never personally seen it falsely detect something - if it is then I would first try re-seating the PIR in the socket and see if that helps. The PIR shouldn't ever trigger unless there is movement.

Feel free to join the discord and I can help you to troubleshoot it easier

oneseventhree, 2023-05-11:
> Hello,
>
> Most likely you need to tune the sensor properly for the room, most of the time users don't realise how sensitive mmWave is because they compare it to a regular PIR. Usually people think [...]

whats the link to the discord :)?

oneseventhree, 2023-05-11:
> Hello,
>
> Most likely you need to tune the sensor properly for the room, most of the time users don't realise how sensitive mmWave is because they compare it to a regular PIR. Usually people think [...]

I never read that document before but thats how I pretty much tuned it myself. I'll give it another go.

The PIR triggers perfectly but its just way too sensitve for the room and triggers as I walk past the door where I can barely see the sensor. Is there a way to tweak the PIR?

I'll join the discord if you can link me to itt :)

closed 2024-07-19 by EverythingSmartHome