EverythingSmartHome/everything-presence-one #64: EP1 pir triggerinmg during the night

umtauscher, 2023-05-12:
Hi, I have a problem lately with 2 differenet EP1s
The PIR triggers several times during the night when nobody is in the room or even awake
This happens frequently, sometimes more than once during one hour.
The mmave stays off at the same time. Maybe this is caused by some updetes of the ESPhome?
(There were several in the last months).
Anyone else having this problem?
Cheers
Wilhem

EverythingSmartHome, 2023-05-12:
Hello,

The PIR is extremely accurate and shouldn't ever trigger falsely. You might try first to remove the PIR from the socket and re-insert it.

Do you have the EP1 facing or near a heat source such as a radiator? PIRs can trigger when a change of heat in it's vision occurs

umtauscher, 2023-05-13:
Well thanks for answering. There is no heat emitting device in the room at night.
Nevertheless I have this phaenomenon on two different EP1s, so I don't think its a hardware issue.
I have reflashed one of them and didn't adopt that one to ESPHome to not get the firmware messed up.
I will see how it goes and will report back.
Cheers

umtauscher, 2023-05-14:
Everything I tried has no effect.
I have more than 20 false positive PIR events during the night.
I am at a loss.
Is there any possibility to reduce the sensitivity of the PIR sensor?

picotrain77, 2023-06-06:
> This happens frequently, sometimes more than once during one hour.
> 
> The mmave stays off at the same time. Maybe this
> 
> Anyone else having this problem?

Yes from the moment I installed them the pir falsely triggers every 5 mins or so randomly. 

I tried placing it in several different positions in the room and even placed it directly where my trusty hue motion sensor has been sitting for years and it still did it.

The only thing that stops it is to point it directly at a wall. Like 2cm away. 

As I cant adjust the sensitivity I have had to regrettably resigned them both as I don't want to use two sensors one for mmWave and one for pir.

EverythingSmartHome, 2023-06-06:
> Yes from the moment I installed them the pir falsely triggers every 5 mins or so randomly.
>
> I tried placing it in several different positions in the room and even placed it directly where my trusty [...]

Hello! Will try to help with this - though Discord might be better for troubleshooting.

Comparing to the Hue motion isn't really a good comparison unfortunately, most Zigbee PIR's are not very sensitive, and the Panasonic PIR included with the EP1 is much more accurate and sensitive than the Hue Motion Sensor.

It's interesting you say pointing at a wall stops the issue from happening, as that would rule out a hardware issue or false detections, since it should continue to happen even when pointing at a wall if that were the case.

Have you tried reseating the sensor in the socket? Where do you have the PIR positioned in the room, what objects does it face?

umtauscher, 2023-07-28:
Just to clarify things:
I have just given up. The thread maybe completed but the problem is not.
Cheers

EverythingSmartHome, 2023-07-28:
> Just to clarify things: I have just given up. The thread maybe completed but the problem is not. Cheers

There was no reply for quite a while so assumed you resolved. By all means I'm willing to help you, just let me know what's going on

umtauscher, 2023-07-28:
sorry, I just haven‘t the time to debug that thing at the moment, I will get back when I have more input.
Thanks

golles, 2023-08-17:
I also see one of the two devices I have does this, only the pir picks up movement while the mm-wave doesn't. It seemed to be starting to do this after I changed the USB-c cable. Will change back to the previous cable and see next night

golles, 2023-08-18:
I've changed to another cable and last night I had no false positives

umtauscher, 2023-08-18:
Thanks, I will try something lime that too. If it‘s supply related, it could be the powersupply as well. Maybe an additional C in the sensor could help as well.

golles, 2023-08-18:
I didn't change the charger, only the cable. I'm using an (old) iPhone charger, because of the low profile.

umtauscher, 2023-08-18:
Yes, I realize that, but since the cable is only used for 5v power, it seems the PIR is critical to ripples on the powerline.

EverythingSmartHome, 2023-08-18:
The PIR is fed 3.3v from the ESP, not attached to the 5v line so I would expect the ESP to have issues also if that were the case - strange problem though, I haven't had any other reports of that but glad you found an easy fix!

umtauscher, 2023-08-19:
it seems to be an interference issue.
I put the powersupply in a direkt wall outlet yesterday and had no false positives through the night. (same powersupply, same cable)
So the PIR seems to be very sensitive to problems that come from the 5v supply. I naturally have no equipment to troubleshoot or record such transients…

golles, 2023-11-24:
Has this issue been resolved for you? For me, this keeps on coming back every now and then on 1 of my 2 devices

umtauscher, 2023-11-24:
No, it is not. I Suspect it's coming from poorly filtered powersuply. 
First I use a noname one. When I switched to one of my apple suplies (iPhone) it got a bit better.
Nevertheless I don't use the EP1 for switching the lights any more, because of it's erratic behaviour.

golles, 2023-11-24:
I've tried different cables and chargers, and I've swapped my 2 devices to see if it could be related to the place, but it keeps on coming back. I've got the impression that reflashing makes it work for a while.

I usually see the pir being active for a longer period than what you see, for me it's ~11 seconds.

[image]
[image]
[image]

I'll now try to swap the PIR sensor from my 2 devices, to see if the problem is with the sensor or with the device itself.

umtauscher, 2023-11-24:
Exchanging the PIRs had no effect on mine.

golles, 2023-11-25:
I can confirm it's not the PIR, got a 10.233s detection during the night on the same unit

umtauscher, 2023-12-05:
I gave up and replaced it with an Aqara FP2 which works flawlessly.

golles, 2023-12-05:
After my last message it has only happened twice. 

I did consider the fp2 as well, but some YouTubers reported ghosting on theirs. 

I'm now considering disabling the pir of the faulty unit.

umtauscher, 2023-12-05:
Well, the EP1 without PIR is just too slow for my taste.