EverythingSmartHome/everything-presence-one #66: Minor inconvenience - Firmware: Unavailable

somejavadev, 2023-05-22:
Hi,

Thanks for all the effort in the creation of the EP1, this is truly an awesome hackable sensor. Something I picked up which I haven't fully figured out why is that the Firmware under the **Configuration** section of home assistant shows 'unavailable' although under the **Device Info** it does show the firmware version:

[image]

If you compare this to another esp32dev I have:

[image]

Here it shows "Up To Date" under the **Configuration** section. I take it home assistant takes the Firmware attribute from the Configuration section to determine if it should alert you for new over the air updates.

The result of this is a minor inconvenience of not being alerted that the esphome firmware is outdated and being able to update it from home assistant. I say it's minor because one can simply update the firmware from ESP Home directly:

[image]

I compared both devices EP1 and a standard esp32dev's esphome yaml but couldn't easily identify what the reason for this could be.  Is this normal for the Firmware attribute to display 'unavailable' and is the expected behaviour or might this be an incorrect configuration on my end?

EverythingSmartHome, 2023-07-27:
Sorry I missed this, is it working now? I note that your config names are different from what the default would be when adopted, did you create these configs manually?

somejavadev, 2023-07-28:
No problem at all. It still seems to be showing Unavailable. I added the sensors via the ESPHome interface using the config provided here: https://github.com/EverythingSmartHome/everything-presence-one/blob/main/everything-presence-one.yaml. 

The steps I followed was:

Go to ESPHome in homeassistant
Create  a new device
Give it a name.
Click Next
Click Skip the installation step

I then pasted the yaml from the main  repo `everything-presence-one.yaml` as the configuration for the new device, I then renamed the values of the substations and gave the device a static IP address as below: 

```yaml
substitutions:
  name: "ep1-one"
  friendly_name: "ep1-one"
  project_name: "Everything Smart Technology.Everything Presence One"
  project_version: "1.1.5"
  temperature_offset: "-3"
  humidity_offset: "5"
  temperature_update_interval: "60s"
  illuminance_update_interval: "30s"
  pir_delay_off: "10s"
  occupancy_delay_off: "15s"
  hidden_ssid: "false"
  factory_reset_disabled: "true"
  uart_target_output_disabled: "true"
  uart_presence_output_disabled: "true"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: ${project_name}
    version: ${project_version}

esp32:
  board: esp32dev

logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  ap: {}
  manual_ip:
    static_ip: 192.168.4.64
    gateway: 192.168.4.1
    subnet: 255.255.255.0

captive_portal:

improv_serial:

esp32_improv:
  authorizer: none

dashboard_import:
  package_import_url: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
  import_full_config: false
[... 274 lines cut ...]
    disabled_by_default: True
    on_press:
      - button.press: restart_mmwave
      - button.press: restart_internal
  - platform: safe_mode
    internal: false
    name: Safe mode
    entity_category: config
    disabled_by_default: True
  - platform: template
    name: Factory reset mmWave
    icon: mdi:cog-counterclockwise
    id: factory_reset_mmwave
    internal: ${factory_reset_disabled}
    entity_category: config
    on_press:
      - switch.turn_off: mmwave_sensor
      - delay: 1s
      - uart.write: "resetCfg"
      - delay: 3s
      - switch.turn_on: mmwave_sensor

```

I then deployed/installed it by letting ESPHome compile the code (legacy format) and installed it with `esp_flasher` via an USB cable.

EverythingSmartHome, 2023-07-28:
That's perhaps the issue is your manually copying and pasting the config and then renaming entities in code (rather than the HA UI) - did you try using the standard way?

fhold, 2023-09-12:
I have the same issue. Interestingly I have 2 EPO Sensors and only one has the problem. 
I just did a fresh firmware install via cable and website (Version ESPHome 2023.7.1) but the problem is the prevails. In Homeassistant it shows unavailable. If I connect via cable and use the web-console you provide via your website I can see it's Version ESPHome 2023.7.1.
The problem is, that no new ESP Update are installed. Would love to fix that somehow

somejavadev, 2023-09-16:
@EverythingSmartHome I have tried the default config and rename the entities in the UI but for some reason the Firmware section remains Unknown. Installation method was as follows:

- Copy the latest release 1.1.6 yaml from github.
- Add a new device in ESPHome within Home Assistant.
- Update the new device in ESPHome's config with the config from github.
- Install via serial.

```
Firmware: 2023.8.3 (Sep 16 2023, 14:25:21)
Hardware: 1.1.6
Home Assistant 2023.9.2
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest
```

 @fhold mine strangely occurs with all three my devices. Did you make any changes to the config you got from github?

fhold, 2023-11-02:
@somejavadev I did not make any changes I used the standard installation process and one is working fine the other one not and I'm not able to do updates on the faulty one. @EverythingSmartHome any idea what we can do?

therealeldaria, 2023-11-30:
It is the same for me, I used the installation from the website.
https://everythingsmarthome.github.io/everything-presence-one/updating.html
I have not done any renaming or anything like that.

I can see the device in Home Assistant but not in the ESPHome addon, and the firmware is showing Unknown.
There was also a recent update to ESPHome and all my ESP's wanted to be updated, but not the EP1, so my thinking is that it is not entirely connected to the rest of the ESPHome eco system.

Not sure if it is related but the EP1 is on a VLAN with no internet access. But I assume it does not need any access to internet.

EverythingSmartHome, 2024-07-19:
Please open a new issue if this is still a problem but will close for now

closed 2024-07-19 by EverythingSmartHome

somejavadev, 2024-10-19:
@EverythingSmartHome  and @therealeldaria This issue is closed but I managed to identify the problem. Under ESPHome I had the default discovery set to "Ping". After enabling mDNS on my router and then disabling ping as the default, the existing devices appeared on the ESPHome interface and had an option to be adopted. After adopting them the firmware versions are now showing correctly and I am able to upgrade them as new versions are released. I believe the problem comes in when having vlans and the vlans don't have mDNS enabled alongside the network home assistant is installed on.