EverythingSmartHome/everything-presence-one #67: Cannot connect EP1 to Home Assistant

briveland1, 2023-05-24:
Hello,

I just received the EP1 device.  I was able to update firmware and connect without a problem using IP address.

Home Assistant is not automatically recognizing device.  I attempted to add manually as per instructions however received error:

"Can't connect to ESP. Please make sure your YAML file contains an 'api:' line."   There is not a yaml file that I can find. 

Thank you

EverythingSmartHome, 2023-05-24:
Hello, that message usually appears if you are entering an incorrect IP address of the EP1 into the box, or Home Assistant isn't able to communicate with that IP, for example firewall rules prevent communication (which might explain why it wasn't auto discovered?)

EverythingSmartHome, 2023-05-25:
Hmm you say you are able to connect via the IP address int the web browser, that shouldn't be possible with the stock firmware as that component was removed.

Did you make sure to select the Home Assistant firmware when you installed it rather than the Smartthings firmware?

briveland1, 2023-05-25:
Yup believe I selected the correct one.   Should I reinstall the firmware to be certain?

briveland1, 2023-05-25:
[image]

briveland1, 2023-05-25:
screen shot of browser view ^^

EverythingSmartHome, 2023-05-25:
That is the smartthings firmware you have selected, not the Home Assistant firmware (you can see the name of the device is everything-presence-st where "st" represents Smartthings.)

briveland1, 2023-05-25:
yup, reinstalling did the trick.   Thank you so much!!!

closed 2023-05-26 by EverythingSmartHome