EverythingSmartHome/everything-presence-one #69: LED Always Enabled After Firmware Update

Veldkornet, 2023-05-27:
Every time my ep1 has a firmware update, the LED is constantly on. I then flip both led switches on, then off and then it goes off. 

As some background, I noticed that every night my ep1 lost WiFi connection when my AP's optimise their channels; but that also caused the presence state to come on when it reconnected to WiFi. 

Therefore, I enabled btm and rrm under the WiFi settings (for roaming). In order to use those settings, I had to switch from adruno to idf... My WiFi disconnect issue is now gone and the device works perfectly, but the LED is always on. 
Actually, I haven't even had my ep1 that long to know if the issue existed with adruno as well....

Here's my full config when using idf though:

’’’
substitutions:
  name: "woonkamer-ep1"
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: !secret api_key

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain
  enable_btm: true
  enable_rrm: true

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

external_components:
  - source: github://pr#3500
    components:
      # list all components modified by this Pull Request here
      - web_server
      - web_server_idf
      - web_server_base
      - captive_portal

’’’

keni-bluebik, 2024-02-05:
My EP1 always turns the LEDs on even after I switch them off in HA. Any suggestions?

EverythingSmartHome, 2024-07-19:
Fixed in newer versions

closed 2024-07-19 by EverythingSmartHome