EverythingSmartHome/everything-presence-one #70: ESPHome reports wrong version of firmware starting with ESPHome v2023.5.2+

devinhedge, 2023-05-31:
This has been a persistent problem.

Steps to recreate:
1. Navigate to ESPHome proper OR to the Update prompt at the top of the HA Settings page
2. Update EP1 via ESPHome OR via the HA Settings page
3. If updating via ESPHome proper, watch EP1 update and notice there are no error messages
4. EP1 successfully updates, and reboots
5. EP1 reports correct version of firmware (currently v2023.5.5) but HA and ESPHome list the current version as v2023.5.2

Is there something that should be updating ESPHome's device catalog or registry with the correct version when EP1 reboots?

devinhedge, 2023-05-31:
[image]

devinhedge, 2023-05-31:
[image]

devinhedge, 2023-05-31:
[image]

It's a bit of a head-scratcher.

elminster13, 2023-05-31:
Mine does the same thing.

Quentame, 2023-06-04:
Rebooting HA solves the issue ?
If yes, the issue is on the HA side

devinhedge, 2023-06-06:
It does not solve the issue. I copied off the last build log and noticed the version is being baked into the build somehow. I haven't looked at the template but I suspect something may be going on there or within any libraries it may be referencing.

norey, 2023-07-06:
How are you able to update its firmware? I'm still on 2023.4.2. I just tried re-flashing it at https://everythingsmarthome.github.io/everything-presence-one/updating.html and the version stayed the same.

[image]

Quentame, 2023-09-20:
I have an EP1 on v`2023.7.1` but the update entity isn't created, so it's not even possible to try to update from HA now (may changed, need a manual update).

Seen the fw version on the device info panel.

brofeelgood, 2023-09-27:
I noticed the same problem previously when I was updating the EP1 firmware wirelessly. For me, updating via wifi works maybe 1 out of 50 times. It would invariably time out before completion. You can see the error in the logs if you perform the update in the ESPHome add-on. If you do it via the HA prompt, it wouldn't show anything and you'd think everything's gone well when in fact it hasn't.

The foolproof way is to do it manually.
1. Go to ESPHome add-on, select the 3 dots > Install > Manual Download > Modern format and save the output file on your PC.
2. Connect EP1 to your PC via USB.
3. Open ESPHome web io (https://web.esphome.io/) on a Chrome browser.
4. Connect and select your EP1 device which should have already been detected.
5. Select Install (not Install for first time), navigate to the file downloaded in step 1 and select it.
6. Start the update and wait (2 minutes).

Anyway, I've stopped updating my EP1 firmware after firmware version 2023.7.1 because that's the version with temp/humidity calibration available in the device page and it basically covers everything I need.

NicoLouis-be, 2023-09-27:
FYI
I added this piece of code, and it shows me the version in diagnostics as a new sensor
so it is able to read the version when you do it that way, don't get it why it's not showing it in the firmware sensor
```
text_sensor:
  - platform: version
    name: ESPHome Version
```
[image]

[image]

norey, 2023-09-27:
> I have an EP1 on v`2023.7.1` but the update entity isn't created, so it's not even possible to try to update from HA now (may changed, need a manual update).
> 
> Seen the fw version on the device info panel.

what info panel?

@brofeelgood 
I only tried updating via https://everythingsmarthome.github.io/everything-presence-one/updating.html and not hass, with EP1 connected via USB.

I'm still on

```
Firmware: 2023.4.2 (Apr 28 2023, 09:31:25)
Hardware: 1.1.3
```

EP1 was integrated with the ESPHome built-in integration. No add-on was involved.

Quentame, 2023-09-27:
> what info panel?
>
>
>
> @brofeelgood
>
> I only tried updating via https://everythingsmarthome.github.io/everything-presence-one/updating.html and not hass, with EP1 connected via USB.
>
>
>
> I'm still on [...]

In this panel:
[image]

Add-on is HA related, not ESPHome.

norey, 2023-09-27:
> In this panel: [image]
>
> Add-on is HA related, not ESPHome.

Yeah, that image is what I posted earlier. Showing the older version.

My main question was about the method of updating EP1 to the latest version. As I mentioned before, it only updated to the 2023.4.2 version.

brofeelgood, 2023-09-27:
> @brofeelgood I only tried updating via https://everythingsmarthome.github.io/everything-presence-one/updating.html and not hass, with EP1 connected via USB.
>
> I'm still on
>
> ```
> Firmware: 2023.4.2 [...]

I'm sorry. I made the silly assumption everyone had the ESPHome add-on installed. Monitoring the logs in the add-on was how I figured out the updates were failing more often than not when done over wifi, regardless of wifi signal strength. Switching over to manual updates was what did the trick for me. I've only ever used the link you provided on initial setup.

A little more info here regarding the ESPHome add-on.

You can find it in Settings > Add-ons.
[image]

The options that are available after clicking on the 3 dots in the add-on.
[image]

What you see in ESPHome web io.
[image]

The steps to update via USB.
[image]

Here's how it looks on my EP1 device page.
[image]

EverythingSmartHome, 2023-09-27:
> I noticed the same problem previously when I was updating the EP1 firmware wirelessly. For me, updating via wifi works maybe 1 out of 50 times. It would invariably time out before completion. You can see the error in the logs if you perform the update in the ESPHome add-on. If you do it via the HA prompt, it wouldn't show anything and you'd think everything's gone well when in fact it hasn't, which is why it'll ask you to update it over and over again.
> 
> The foolproof way is to do it manually.
> 
> 1. Go to ESPHome add-on, select the 3 dots > Install > Manual Download > Modern format and save the output file on your PC.
> 2. Connect EP1 to your PC via USB.
> 3. Open ESPHome web io (https://web.esphome.io/) on a Chrome browser.
> 4. Connect and select your EP1 device which should have already been detected.
> 5. Select Install (not Install for first time), navigate to the file downloaded in step 1 and select it.
> 6. Start the update and wait (2 minutes).
> 
> Anyway, I've stopped updating my EP1 firmware after version 2023.7.1 because that's the version with temp/humidity calibration available in the device page and that basically covers everything I need.

There is no need to update this way, you can do it over WiFi but you need to press the "Safe Mode" button  from the device page first - this shuts down most components and allows the WiFi update to complete. Hope that helps

brofeelgood, 2023-09-27:
Thanks for the clarification Lewis. I always wondered what that button was about. The first (and only) time I clicked it, it wiped out all my settings and I had to set up everything again. Is that normal/expected behaviour?

EverythingSmartHome, 2023-09-27:
> Thanks for the clarification Lewis. I always wondered what that button was about. The first (and only) time I clicked it, it wiped out all my settings and I had to set up everything again. Is that normal/expected behaviour?

It doesn't wipe out your settings but it will make everything "unavailable" until the device is restarted or flashed. All your settings and everything remain untouched. The safe mode button shuts down most components (that's why they go unavailable) so that the ESP32 doesn't get interrupted when it's doing the update over WiFi

brofeelgood, 2023-09-28:
I'm happy to report everything made it in one attempt (via WiFi) by initiating "Safe Mode" before updating just as Lewis advised. The only minor glitch was "Safe Mode" getting cancelled halfway through the esp32 image compilation, but I quickly reactivated it and everything came through just fine, cheers.

norey, 2024-01-07:
How do I update via wifi? I'm still confused. I see the "Safe Mode" button in the device page. What needs to be done after pressing it?

That's what I see in the device page:

[image]

closed 2024-07-19 by EverythingSmartHome