EverythingSmartHome/everything-presence-one #74: Modifiable properties for temperature / humidity offset

liquidbrains, 2023-06-26:
Hi.  

According to the documentation, via the Samsung integration, one can adjust the offsets for temp / humidity. I cannot find such properties on the HASS integration.  I think one can set that in substitutions, but that requires a rebuild / deploy.  

Would it be possible to have this as a property on the device configuration?

EverythingSmartHome, 2023-06-26:
It's unfortunately possible at the moment as ESPHome does not support this, that's why the substitutions is required. Redeploying doesn't take too long so should be OK!

liquidbrains, 2023-06-26:
Thanks for the reply.  Would it be possible to add documentation for the offset substitutions?  I'm still not certain I've done it correctly or if the weather has just changed and now it's reading a lower temperature.

jfroy, 2023-06-27:
I have been experimenting with dynamic offsets using a template number and a lambda filter. Seems to work.

```yaml
sensor:
  - platform: shtcx
    temperature:
      name: "${room} Temperature"
      filters:
        - lambda: return x + id(temperature_offset).state;
    humidity:
      name: "${room} Humidity"
      filters:
        - lambda: return x + id(humidity_offset).state;
    address: 0x70
    update_interval: ${temperature_update_interval}

number:
  - platform: template
    name: Temperature offset
    id: temperature_offset
    entity_category: config
    min_value: -100.0
    max_value: 100.0
    initial_value: 0.0
    optimistic: true
    step: 0.1
    restore_value: true
    unit_of_measurement: "°C"
```

gbutts, 2023-07-08:
I concur with @liquidbrains. As it seems that temperature compensation is likely to be commonly required for individual setups, can you please add some documentation that walks us through the process. 

I have been unable to work how to adjust for the +5 error I am seeing. (A node-red formula work around is my current option)

EverythingSmartHome, 2023-07-26:
> I have been experimenting with dynamic offsets using a template number and a lambda filter. Seems to work.
> 
> ```yaml
> sensor:
>   - platform: shtcx
>     temperature:
>       name: "${room} Temperature"
>       filters:
>         - lambda: return x + id(temperature_offset).state;
>     humidity:
>       name: "${room} Humidity"
>       filters:
>         - lambda: return x + id(humidity_offset).state;
>     address: 0x70
>     update_interval: ${temperature_update_interval}
> 
> number:
>   - platform: template
>     name: Temperature offset
>     id: temperature_offset
>     entity_category: config
>     min_value: -100.0
>     max_value: 100.0
>     initial_value: 0.0
>     optimistic: true
>     step: 0.1
>     restore_value: true
>     unit_of_measurement: "°C"
>   - platform: template
>     name: Humidity offset
>     id: humidity_offset
>     entity_category: config
>     min_value: -100.0
>     max_value: 100.0
>     initial_value: 0.0
>     optimistic: true
>     step: 0.1
>     restore_value: true
>     unit_of_measurement: "%"
> ```

Sorry for the delay, thanks for this I am going to try it now and make a firmware soon! This didn't used to work I don't think so really cool it now does!

liquidbrains, 2023-07-26:
Do you know if the use of lambdas could make the board respond slower?  I don't know if I'm imagining it, but it feels like it's taking longer to trigger occupancy and that's the only thing I can think of.

EverythingSmartHome, 2023-07-26:
I'm not seeing any perceivable slowdowns on the unit I'm testing on personally

EverythingSmartHome, 2023-07-27:
Fixed by https://github.com/EverythingSmartHome/everything-presence-one/pull/89