EverythingSmartHome/everything-presence-one #75: mDNS lookup failure

TomHart, 2023-06-28:
My EP1 mDNS lookup fails, meaning ESPHome thinks the device is offline. Other ESP devices are fine. I haven't changed anything other than the temp offset, can you suggest anything to try? I've tried removing my DHCP lease, and unplugging/replugging the device in, so avail.

```bash
root@TOM-PC:/mnt/c/Windows/system32# dig esp-cam-1 +short
192.168.5.233
root@TOM-PC:/mnt/c/Windows/system32# dig esp-cam-1.lan +short
192.168.5.233
root@TOM-PC:/mnt/c/Windows/system32# dig esp-cam-1.local +short
192.168.5.233
root@TOM-PC:/mnt/c/Windows/system32# dig everything-presence-one-707ec8 +short
192.168.1.98
root@TOM-PC:/mnt/c/Windows/system32# dig everything-presence-one-707ec8.lan +short
192.168.1.98
root@TOM-PC:/mnt/c/Windows/system32# dig everything-presence-one-707ec8.local +short
root@TOM-PC:/mnt/c/Windows/system32#
```

TomHart, 2023-12-21:
Still having this issue, has anyone else noticed it too?

EverythingSmartHome, 2024-07-19:
So sorry for not getting back to you, somehow I missed this issue. This would be more of a ticket for ESPHome as the EP1 doesn't do anything special here - if it is still an issue for you.

closed 2024-07-19 by EverythingSmartHome

TomHart, 2024-08-22:
No worries! Cheers :)