EverythingSmartHome/everything-presence-one #80: Breaking change introduced in a minor version number change

Ongion, 2023-07-07:
The latest update (1.1.3 -> 1.1.4) changed the names of every single sensor. This means I have to update every single automation or panel in Home Assistant that refers to the old names.

This kind of breaking change should not have been made as a minor version change, especially without some kind of announcement. I got the update automatically while just trying to change a temperature offset, and wondered why my lights stopped turning on.

Can this at least be a major semver update (1.2.0)?

Ongion, 2023-07-07:
Looks like the new substitution is using "Friendly Name". I still feel this should at least be a major version change (1.1.3 -> 1.2.0 instead of 1.1.4), since it breaks existing configurations.

TNTPro, 2023-07-09:
This is definitely a small change that is having big effects.  I have added friendly names in my device YAMLs, which has changed the Names of the sensors, but the IDs are still named as "sensor.everything_presence_one_humidity_2" etc.  Also, to add to the confusion, the devces don't have matching numbers i.e. a single EP1 might have "sensor.everything_presence_one_humidity_2" and "sensor.everything_presence_one_illuminance_3"

TNTPro, 2023-07-09:
I edited the "everything-presence-one.yaml" file and from ESPHome, Installed the EP1s again and all seems good for me.

neildsb, 2023-07-10:
Please can you share a before and after YAML, this is a breaking change for me :-(

brofeelgood, 2023-07-10:
> Please can you share a before and after YAML, this is a breaking change for me :-( I am using ESPHome only

**Go to ESPHome (add-on) -> Edit -> replace the substitution "room" with "friendly_name" -> Save -> Install.**

The "room" substitution is not the only thing messed up in the 2023.6.5 update, so are the entities for "Firmware", "mmWave LED" and "Restart everything-presence-one-XXXXXX". They're greyed out and when clicked on, show "This entity is no longer being provided by the esphome integration. If the entity is no longer in use, delete it in settings.

[image]

neildsb, 2023-07-10:
> **Go to ESPHome (add-on) -> Edit -> replace the substitution "room" with "friendly_name" -> Save -> Install.**
>
> The "room" substitution is not the only thing messed up in the 2023.6.5 update, so are [...]

thanks, I needed to delete and re-add to HA

TNTPro, 2023-07-10:
[everything-presence-one.yaml.1.1.4.TEC.zip](https://github.com/EverythingSmartHome/everything-presence-one/files/12005158/everything-presence-one.yaml.1.1.4.TEC.zip)
Here are a few versions

TNTPro, 2023-07-10:
> **Go to ESPHome (add-on) -> Edit -> replace the substitution "room" with "friendly_name" -> Save -> Install.**
>
> The "room" substitution is not the only thing messed up in the 2023.6.5 update, so are [...]

These items have been renamed and moved and or disabled.  So you can delete the old ones and enable the new ones (if you need them enabled)

TNTPro, 2023-07-10:
> 
> thanks, I needed to delete and re-add to HA
These old entities become deletable after a HA restart.

Menz01, 2023-07-18:
> > thanks, I needed to delete and re-add to HA
> 
> These old entities become deletable after a HA restart.

maybe you or someone else can help me? i have an entity that was created cause of this issue that i cannot delete and i would like to make it gone
[image]

Ongion, 2023-07-18:
> You are receiving this email because you authored the thread.

I think this entity is still provided by the device, it's just disabled.
You can't delete it because, well, it still exists.

Menz01, 2023-07-19:
> Try this:
> 
> 1. Enable the entity, then delete.
> 2. If enabling fails, do a full HA restart, then delete.

THANK YOU!!!! #2 is what did it but #1 is also necessary. my OCD thanks you!!

EverythingSmartHome, 2023-07-19:
Sorry guys, this was an unintentional change - it was only intended to affect new users, didn't realise you would get prompted to update which would cause this issue.

Will be more careful in the future!

liquidbrains, 2023-07-23:
> > Try this:
> > 
> > 1. Enable the entity, then delete.
> > 2. If enabling fails, do a full HA restart, then delete.
> 
> THANK YOU!!!! #2 is what did it but #1 is also necessary. my OCD thanks you!!

I don't know how I managed it but I had deleted so many things (from HASS and ESPHome), restarted, deleted again, renamed the remaining ESPHome entity to wookie (because aaaaahrg).  Then I disabled ESPHome while I flashed it cleanly from https://everythingsmarthome.github.io/everything-presence-one/.  Then when I added it back to ESPHome, it "adopted" and reflashed the device again, after which my old "wookie" device came back to life, but this time with sensors being published to the UI.  I took the chance and renamed it to the correct name and ... It seems to have sticked?!?   I'm never touching it again!

jokerigno, 2023-10-27:
Hi, sorry just to clarify. I faced this issue and even if my config has "room" and "friendly_name" all my devices sensor are called "binary_sensor.everything_presence_one_2707bxxx_xxx" instead of "binary_sensor.[friendly_name]_xx".

So it's a issue or a breaking change? Just to understand if I have to rename all my automation or wait for a fix.

closed 2024-07-19 by EverythingSmartHome