EverythingSmartHome/everything-presence-one #82: EP1 Sensor red led operation

paul-nzl, 2023-07-15:
Newish Home Assistant user and First time user of ESPHome
I have tried two Everything Presence One EP1 sensors with the same result for both as show in the table below.
The  Humidity, Illuminance,  mmWave, Occupancy, PIR and Temperature as normal.
The issue i have is with the operation of the ESPHome EP1 mmWaveSensor Switch, ESPHome EP1 mmWaveLed Switch and how they interact with the Red Sensor Led.
Why is the sensor red led on when the  mmWaveSensor and mmWaveLed switches on the dashboard are off.
This makes no sense to me, where am i going wrong.

|ESPHomeDashboard| ESPHomeDashboard |Sensor||
|-|-|-|-|		
|mmWaveSwitch|	mmWaveLedSwitch	|	Led|		Detection|
|Off	|Off |On| No|
|On	|On	| Flashing| Yes|
|Off	| On	| On	| No|
|On	| Off	| Off| Yes|

ESPHome Current version: 2023.6.5 

Home Assistant 2023.7.2
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest

Raspberry Pi 4
4GB
WD SATA 240 GB SSD 
Disk Use 5%

INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/everything-presence-one-6ea030.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from everything-presence-one-6ea030.local using esphome API
INFO Successfully connected to everything-presence-one-6ea030.local
[10:58:58][I][app:102]: ESPHome version 2023.6.5 compiled on Jul 13 2023, 13:06:52
[10:58:58][I][app:104]: Project Everything_Smart_Technology.Everything_Presence_One version 1.1.4
[10:58:58][C][status_led:064]: Status Led Light:
[10:58:58][C][status_led:065]:   Pin: GPIO32
[10:58:58][C][wifi:543]: WiFi:
[10:58:58][C][wifi:379]:   Local MAC: '[redacted]
[10:58:58][C][wifi:380]:   SSID: '[redacted]
[10:58:58][C][wifi:381]:   IP Address: 192.168.1.80
[10:58:58][C][wifi:383]:   BSSID: '[redacted]
[10:58:58][C][wifi:384]:   Hostname: 'everything-presence-one-6ea030'
[10:58:58][C][wifi:386]:   Signal strength: -54 dB ▂▄▆█
[10:58:58][C][wifi:390]:   Channel: 1
[10:58:58][C][wifi:391]:   Subnet: 255.255.255.0
[10:58:58][C][wifi:392]:   Gateway: 192.168.1.254
[10:58:58][C][wifi:393]:   DNS1: 192.168.1.254
[10:58:58][C][wifi:394]:   DNS2: 0.0.0.0
[10:58:58][C][logger:301]: Logger:
[10:58:58][C][logger:302]:   Level: DEBUG
[10:58:58][C][logger:303]:   Log Baud Rate: 115200
[10:58:58][C][logger:305]:   Hardware UART: UART0
[10:58:58][C][i2c.arduino:053]: I2C Bus:
[10:58:58][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[10:58:58][I][i2c.arduino:069]: Results from i2c bus scan:
[10:58:58][I][i2c.arduino:075]: Found i2c device at address 0x23
[10:58:58][I][i2c.arduino:075]: Found i2c device at address 0x70
[10:58:58][C][uart.arduino_esp32:124]: UART Bus 1:
[10:58:58][C][uart.arduino_esp32:125]:   TX Pin: GPIO13
[10:58:58][C][uart.arduino_esp32:126]:   RX Pin: GPIO14
[10:58:58][C][uart.arduino_esp32:128]:   RX Buffer Size: 256
[10:58:58][C][uart.arduino_esp32:130]:   Baud Rate: 115200 baud
[10:58:58][C][uart.arduino_esp32:131]:   Data Bits: 8
[10:58:58][C][uart.arduino_esp32:132]:   Parity: NONE
[10:58:58][C][uart.arduino_esp32:133]:   Stop bits: 1
[10:58:58][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'mmWave'
[10:58:58][C][gpio.binary_sensor:015]:   Device Class: 'occupancy'
[10:58:58][C][gpio.binary_sensor:016]:   Pin: GPIO15
[... 457 lines cut ...]
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[13:54:35] INFO: Configuring timezone (Pacific/Auckland)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service discovery: starting
[13:54:36] INFO: Waiting for ESPHome dashboard to come up...
[13:54:37] INFO: Starting ESPHome dashboard...
2023-07-15 13:54:38,032 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[13:54:38] INFO: Starting NGINX...
[13:54:39] INFO: Successfully send discovery information to Home Assistant.
s6-rc: info: service discovery successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2023-07-15 13:55:57,115 INFO 200 GET /devices (0.0.0.0) 6.80ms

EverythingSmartHome, 2023-07-15:
Unfortunately the firmware on the mmWave sensor board dictates how the LED works that can't be overridden. For example, if you turn the LED off, then switch off the sensor and turn it back on, that will re-enable the LED automatically. Frustrating but unfortunately the way it is.

What were you hoping to achieve?

paul-nzl, 2023-07-15:
Thanks for clarification, I assumed something was setup wrong.
I was trying to achieve.  Sensor led Off and detection Off

EverythingSmartHome, 2023-07-15:
Be aware when changing settings frequently (i.e daily) that use flash memory on the sensor (mmWave on/off, led, distance, sensitivity), it may cause a degradation in the flash memory as it uses a type that is only rated for a certain amount of cycles (we don't know how many as it's not published) - the settings are not designed to be changed on a daily basis

closed 2023-07-19 by EverythingSmartHome