EverythingSmartHome/everything-presence-one #83: ESPHome 2023.7.0b2 has a breaking Change For Switches - restore_state changed to restore_mode and no longer boolean

jweston2112, 2023-07-17:
```yaml

INFO ESPHome 2023.7.0b2
INFO Reading configuration /config/esphome/everything-presence-one-d531b0.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

switch.template: [source /config/esphome/.esphome/packages/f43db209/everything-presence-one.yaml:150]
  platform: template
  name: mmWave sensor
  id: mmwave_sensor
  disabled_by_default: True
  entity_category: config
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  turn_on_action: 
    - uart.write: sensorStart
    - delay: 1s
  turn_off_action: 
    - uart.write: sensorStop
    - delay: 1s
switch.template: [source /config/esphome/.esphome/packages/f43db209/everything-presence-one.yaml:164]
  platform: template
  name: UART presence output
  id: uart_presence_output
  entity_category: config
  internal: true
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 1 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
switch.template: [source /config/esphome/.esphome/packages/f43db209/everything-presence-one.yaml:188]
  platform: template
  name: UART target output
  id: uart_target_output
  entity_category: config
  internal: true
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  assumed_state: False
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 2 1 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 2 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
```

neildsb, 2023-07-19:
+1 with 2023.7.0

Home Assistant 2023.7.2
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest

INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/everything-presence-one-1.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

switch.template: [source /config/esphome/.esphome/packages/0c0347a4/everything-presence-one.yaml:150]
  platform: template
  name: mmWave sensor
  id: mmwave_sensor
  disabled_by_default: True
  entity_category: config
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  turn_on_action: 
    - uart.write: sensorStart
    - delay: 1s
  turn_off_action: 
    - uart.write: sensorStop
    - delay: 1s
switch.template: [source /config/esphome/.esphome/packages/0c0347a4/everything-presence-one.yaml:164]
  platform: template
  name: UART presence output
  id: uart_presence_output
  entity_category: config
  internal: true
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 1 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
switch.template: [source /config/esphome/.esphome/packages/0c0347a4/everything-presence-one.yaml:188]
  platform: template
  name: UART target output
  id: uart_target_output
  entity_category: config
  internal: true
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  assumed_state: False
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 2 1 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 2 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor

EverythingSmartHome, 2023-07-19:
Thanks, pushed fix to the main YAML now!

closed 2023-07-19 by EverythingSmartHome

EverythingSmartHome, 2023-07-19:
Only you can answer that, the choice is yours as with every update that shows up 😅

neildsb, 2023-07-19:
cool beans, I have just tried again now there is a fix, still getting the same errors :-(

jefutte, 2023-07-19:
@neildsb I had to remove the files in /config/esphome/.esphome/packages/<package id>

Seems it's caching the files, and don't know when to pull a new version from Github.

neildsb, 2023-07-19:
Nice, for others I renamed/attached the Dir with temp, just to be safe

[image]

started the upgrade again, so far, so good!