EverythingSmartHome/everything-presence-one #85: Changes in ESPHome 2023.7.0 produce 'failed config'

clfberlin, 2023-07-19:
Hi. The EP1 reports "failed config" after ESPHome 2023.7.0 update.

```
Failed config

switch.template: [source /config/esphome/.esphome/packages/0c0347a4/everything-presence-one.yaml:150]
  platform: template
  name: mmWave sensor
  id: mmwave_sensor
  disabled_by_default: True
  entity_category: config
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  turn_on_action: 
    - uart.write: sensorStart
    - delay: 1s
  turn_off_action: 
    - uart.write: sensorStop
    - delay: 1s
switch.template: [source /config/esphome/.esphome/packages/0c0347a4/everything-presence-one.yaml:164]
  platform: template
  name: UART presence output
  id: uart_presence_output
  entity_category: config
  internal: true
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 1 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
switch.template: [source /config/esphome/.esphome/packages/0c0347a4/everything-presence-one.yaml:188]
  platform: template
  name: UART target output
  id: uart_target_output
  entity_category: config
  internal: true
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  assumed_state: False
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 2 1 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setUartOutput 2 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor

```

clfberlin, 2023-07-19:
It works, but it wasn't the cache. It only worked when I physically removed the `@main` from the github link in the yaml file.

jlips24, 2023-07-20:
Thanks @clfberlin! Removing `@main` seemed to do the trick.