EverythingSmartHome/everything-presence-one #87: Can't connect to ESP. Please make sure your YAML file contains an 'api:' line.

cchurch132, 2023-07-24:
First off, go easy, I'm very green when it comes to this stuff, but I'm trying to connect my Everything Presence One to HA and I get the error "Can't connect to ESP. Please make sure your YAML file contains an 'api:' line." when following the auto method listed in the docs. I've followed the integration steps well up to this point.

EverythingSmartHome, 2023-07-24:
Did home assistant automatically discover the EP1 in the integrations menu or are you adding it by IP address?

cchurch132, 2023-07-24:
> Did home assistant automatically discover the EP1 in the integrations menu or are you adding it by IP address?

oh, forgot that part. HA discovered it automatically.

EverythingSmartHome, 2023-07-24:
Can you share a screenshot of the auto discovered integration and each step your going through to the error?

cchurch132, 2023-07-24:
Sure thing. After following the docs and getting to this point (https://everythingsmarthome.github.io/everything-presence-one/Home%20Assistant/connecting-home-assistant.html), I've logged into HA and see the sensor has been discovered. Then following these steps, waiting about 1-2 mins after clicking Submit, I get the error mentioned. 
[image]
[image]
[image]
[image]

EverythingSmartHome, 2023-07-24:
Have you confirmed that IP address is definitely correct? The symptoms your getting are what happens if an incorrect address is added.

You can confirm the address by going to the user guide firmware page from before, click connect and select the USB port. Instead of hitting install this time, hit logs. In the bottom right, hit reset device, give the device 10 seconds or so to boot, then copy and paste the logs here

cchurch132, 2023-07-24:
Confirmed IP address in HA matches the listing on my network. 

Here's the logs before the reset
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:262]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][esp32_ble:027]: Setting up BLE...
[D][esp32_ble:043]: BLE setup complete
[D][esp32_ble_server:034]: Setting up BLE Server...
[D][esp32_ble_server:071]: BLE server setup successfully
[C][wifi:038]: Setting up WiFi...
[C][wifi:048]: Starting WiFi...
[C][wifi:049]:   Local MAC: A0:B7:65:FA:27:FC
[D][wifi:058]: Loaded saved wifi settings: UniFi_IOT
[D][wifi:425]: Starting scan...
[D][esp32_improv.component:201]: Setting Improv to start
[D][esp32_improv.component:076]: Service started!
[D][wifi:440]: Found networks:
[I][wifi:484]: - 'UniFi_IOT' (D2:21:F9:9F:E3:2D) [redacted]▂▄▆█
[D][wifi:485]:     Channel: 11
[D][wifi:486]:     RSSI: -22 dB
[I][wifi:484]: - 'UniFi_IOT' (72:A7:41:9B:33:C6) [redacted]▂▄▆█
[D][wifi:485]:     Channel: 11
[D][wifi:486]:     RSSI: -67 dB
[I][wifi:484]: - 'UniFi_IOT' (D2:21:F9:9F:FE:C9) [redacted]▂▄▆█
[D][wifi:485]:     Channel: 1
[D][wifi:486]:     RSSI: -71 dB
[I][wifi:484]: - 'UniFi_IOT' (D2:21:F9:9F:D8:2D) [redacted]▂▄▆█
[D][wifi:485]:     Channel: 1
[D][wifi:486]:     RSSI: -83 dB
[I][wifi:484]: - 'UniFi_IOT' (76:A7:41:EC:B7:C8) [redacted]▂▄▆█
[D][wifi:485]:     Channel: 11
[D][wifi:486]:     RSSI: -85 dB
[D][wifi:489]: - ''[redacted] (D2:21:F9:CF:E3:2D) [redacted]▂▄▆█
[D][wifi:489]: - 'UniFi'[redacted] (D0:21:F9:EF:E3:2D) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:BF:E3:2D) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:AF:E3:2D) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:44:0A:78) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:34:0A:78) [redacted]▂▄▆█
[D][wifi:489]: - 'UniFi'[redacted] (D0:21:F9:54:0A:78) [redacted]▂▄▆█
[D][wifi:489]: - 'UniFi'[redacted] (70:A7:41:9B:33:C6) [redacted]▂▄▆█
[D][wifi:489]: - 'WiFi-0C34'[redacted] (E8:48:B8:76:0C:34) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (72:A7:41:BB:33:C6) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (72:A7:41:AB:33:C6) [redacted]▂▄▆█
[D][wifi:489]: - 'UniFi'[redacted] (D0:21:F9:EF:FE:C9) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:BF:FE:C9) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (72:A7:41:CB:33:C6) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:CF:FE:C9) [redacted]▂▄▆█
[D][wifi:489]: - 'ORBILEIJ'[redacted] (34:98:B5:34:7B:13) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (3A:98:B5:34:7B:13) [redacted]▂▄▆█
[D][wifi:489]: - 'TelstraA6CB02'[redacted] (BC:30:D9:A6:CB:04) [redacted]▂▄▆█
[D][wifi:489]: - 'OPTUS_A32E6DM'[redacted] (3C:58:5D:A3:2E:6F) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:BF:D8:2D) [redacted]▂▄▆█
[D][wifi:489]: - 'UniFi'[redacted] (D0:21:F9:EF:D8:2D) [redacted]▂▄▆█
[D][wifi:489]: - 'Harycees'[redacted] (3C:58:5D:C0:D0:02) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (7E:A7:41:EC:B7:C8) [redacted]▂▄▆█
[D][wifi:489]: - 'Bloodrose '[redacted] (28:87:BA:46:2E:DC) [redacted]▂▄▆█
[D][wifi:489]: - ''[redacted] (D2:21:F9:AF:D8:2D) [redacted]▂▄▆█
[... 37 lines cut ...]
[C][wifi:391]:   Subnet: 255.255.255.0
[C][wifi:392]:   Gateway: 192.168.107.1
[C][wifi:393]:   DNS1: 192.168.107.1
[C][wifi:394]:   DNS2: 0.0.0.0
[C][logger:301]: Logger:
[C][logger:302]:   Level: DEBUG
[C][logger:303]:   Log Baud Rate: 115200
[C][logger:305]:   Hardware UART: UART0
[C][esp32_ble:238]: ESP32 BLE:
[C][esp32_ble:240]:   MAC address: A0:B7:65:FA:27:FE
[C][esp32_ble:241]:   IO Capability: none
[C][esp32_ble_server:161]: ESP32 BLE Server:
[C][esp32_improv.component:215]: ESP32 Improv:
[C][esp32_improv.component:217]:   Status Indicator: 'NO'
[C][captive_portal:088]: Captive Portal:
[C][web_server:161]: Web Server:
[C][web_server:162]:   Address: esphome-web.local:80
[C][mdns:112]: mDNS:
[C][mdns:113]:   Hostname: esphome-web-fa27fc
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: esphome-web.local:3232
[C][api:138]: API Server:
[C][api:139]:   Address: esphome-web.local:6053
[C][api:143]:   Using noise encryption: NO
[C][improv_serial:032]: Improv Serial:

EverythingSmartHome, 2023-07-24:
The firmware you have on this device is not the EP1 firmware from the user guide, you can see from the logs the hostname is: esphome-web-fa27fc, where as if the ep1 firmware is loaded it starts "everything-presence-one-fa27fc"

It looks like this firmware has been flashed from the ESPHome web tools site? Firmware needs to be flashed from the user guide first

cchurch132, 2023-07-24:
I added ESPHome to HA and logged in via the UI after I first got the error. I'm going through the start up guide to start fresh now

cchurch132, 2023-07-24:
Ok, started from scratch. Still getting the same error. Haven't used ESPHome tools at all. 

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:262]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:183]: Performing I2C bus recovery
[C][uart.arduino_esp32:077]: Setting up UART...
[C][status_led:055]: Setting up Status LED...
[D][binary_sensor:034]: 'mmWave': Sending initial state OFF
[D][template.switch:046]:   Restored state ON
[D][switch:012]: 'mmWave sensor' Turning ON.
[D][switch:055]: 'mmWave sensor': Sending state ON
[D][template.switch:046]:   Restored state OFF
[D][switch:016]: 'UART presence output' Turning OFF.
[D][switch:016]: 'mmWave sensor' Turning OFF.
[D][switch:055]: 'mmWave sensor': Sending state OFF
[D][switch:055]: 'UART presence output': Sending state OFF
[D][template.switch:046]:   Restored state OFF
[D][switch:016]: 'UART target output' Turning OFF.
[D][switch:016]: 'mmWave sensor' Turning OFF.
[D][switch:055]: 'UART target output': Sending state OFF
[D][number:012]: 'mmWave distance': Sending state 315.000000
[D][number:012]: 'mmWave off latency': Sending state 15.000000
[D][number:012]: 'mmWave on latency': Sending state 0.000000
[D][number:012]: 'mmWave sensitivity': Sending state 7.000000
[C][light:035]: Setting up light 'ESP32 status LED'...
[D][status_led:038]: 'ESP32 status LED': Setting initial state
[D][status_led:049]: 'ESP32 status LED': Setting state OFF
[D][light:036]: 'ESP32 status LED' Setting:
[D][light:041]:   Color mode: 
[C][light:035]: Setting up light 'mmWave LED'...
[D][light:036]: 'mmWave LED' Setting:
[D][light:041]:   Color mode: 
[C][shtcx:028]: Setting up SHTCx...
[C][shtcx:056]:   Device identified: SHTC3 (0887)
[C][bh1750.sensor:041]: Setting up BH1750 'Illuminance'...
[C][esp32_ble:027]: Setting up BLE...
[D][esp32_ble:043]: BLE setup complete
[D][esp32_ble_server:034]: Setting up BLE Server...
[D][esp32.preferences:114]: Saving 2 preferences to flash...
[D][esp32.preferences:143]: Saving 2 preferences to flash: 1 cached, 1 written, 0 failed
[D][status_led:029]: Restoring light state OFF
[D][switch:016]: 'mmWave sensor' Turning OFF.
[D][uart_debug:158]: >>> "sensorStartsensorStopsensorStopsensorStop"
[D][uart_debug:158]: <<< "sensorStart\r\n"
[D][uart_debug:158]: <<< "Done\r\n"
[D][uart_debug:158]: <<< "leapMMW:/>sensorStop\r\n"
[D][uart_debug:158]: <<< "Done\r\n"
[D][shtcx:100]: Got temperature=24.01°C humidity=36.92%
[D][sensor:094]: 'Temperature': Sending state 21.00558 °C with 1 decimals of accuracy
[D][sensor:094]: 'Humidity': Sending state 41.92474 % with 1 decimals of accuracy
[D][esp32_ble_server:071]: BLE server setup successfully
[C][wifi:038]: Setting up WiFi...
[C][wifi:048]: Starting WiFi...
[C][wifi:049]:   Local MAC: A0:B7:65:FA:27:FC
[D][wifi:058]: Loaded saved wifi settings: UniFi_IOT
[D][wifi:425]: Starting scan...
[D][esp32_improv.component:201]: Setting Improv to start
[... 246 lines cut ...]
[D][sensor:094]: 'Temperature': Sending state 20.97888 °C with 1 decimals of accuracy
[D][sensor:094]: 'Humidity': Sending state 41.89423 % with 1 decimals of accuracy
[D][binary_sensor:036]: 'PIR': Sending state ON
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=18.1lx
[D][sensor:094]: 'Illuminance': Sending state 18.11024 lx with 1 decimals of accuracy
[D][binary_sensor:036]: 'PIR': Sending state OFF
[D][binary_sensor:036]: 'PIR': Sending state ON
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=18.1lx
[D][sensor:094]: 'Illuminance': Sending state 18.11024 lx with 1 decimals of accuracy
[D][shtcx:100]: Got temperature=24.02°C humidity=36.91%
[D][sensor:094]: 'Temperature': Sending state 21.02161 °C with 1 decimals of accuracy
[D][sensor:094]: 'Humidity': Sending state 41.90796 % with 1 decimals of accuracy
[D][binary_sensor:036]: 'PIR': Sending state OFF
[D][binary_sensor:036]: 'PIR': Sending state ON
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=18.1lx
[D][sensor:094]: 'Illuminance': Sending state 18.11024 lx with 1 decimals of accuracy
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=18.3lx
[D][sensor:094]: 'Illuminance': Sending state 18.33661 lx with 1 decimals of accuracy
[D][shtcx:100]: Got temperature=23.99°C humidity=36.94%
[D][sensor:094]: 'Temperature': Sending state 20.99490 °C with 1 decimals of accuracy
[D][sensor:094]: 'Humidity': Sending state 41.94153 % with 1 decimals of accuracy
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=18.1lx
[D][sensor:094]: 'Illuminance': Sending state 18.11024 lx with 1 decimals of accuracy
[D][binary_sensor:036]: 'PIR': Sending state OFF
[D][binary_sensor:036]: 'PIR': Sending state ON

cchurch132, 2023-07-25:
> Those logs are now correct and you have now loaded the correct firmware.
> 
> I note you have an IOT SSID, is communication and firewall rules setup correctly between VLANs?

*Facepalm* I do have firewall rules set across the vlans, but putting the EP1 on the same VLAN as HA did the trick. I'm so sorry to bother you with this, though i really appreciate your assistance!! Can't wait to get started with it.

closed 2023-07-27 by EverythingSmartHome