EverythingSmartHome/everything-presence-one #90: Not all Configuration entities are available

Iraijus, 2023-07-27:
Greetings. When going through the documentation, I noticed that I should have a few more configuration options available to me than I currently do.
[image]

I have reinstalled the firmware a few times and am currently running the beta firmware, but I did have the exact same problem with the most recent stable version.

EverythingSmartHome, 2023-07-27:
Hello, which options are you looking for that are missing?

Iraijus, 2023-07-27:
Thank you for the quick response! As you can see in the screenshot, there are 5 entities that are unavailable. This is what I meant in my previous message. They do show up, but the HA entities are unavailable, so I cannot configure/use them. Are they supposed to be available, or did I misunderstand this part?

EverythingSmartHome, 2023-07-27:
Ah, this is just a Home Assistant feature. Simply click on the entity that is unavailable, click settings and then enable it and press save.

Iraijus, 2023-07-27:
Now I feel a bit stupid. I knew about the disabling feature, of course, but I have never seen auto-disabled entities that were not disabled because of problems with my devices. Good to know that this can happen. Thank you!

closed 2023-07-27 by Iraijus