EverythingSmartHome/everything-presence-one #96: Cannot flash Beta firmware from Home Assistant ESPHome

liquidbrains, 2023-08-01:
Hi.

I think I might not be using the correct method, but when I try to flash the beta firmware using 
`...
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one-beta.yaml@main
...`

I get the following error:
`    Could not find file '/config/esphome/SEN0395_distance.h'. Please make sure it exists (full path: /config/esphome/SEN0395_distance.h).`

Is ESPHome perhaps misguided in the path of the header file?

EverythingSmartHome, 2023-08-01:
I would flash via the cable first using the user guide and delete and re-adopt the config in ESPHome before trying OTA

pyrodex, 2023-08-02:
> Adopting step after flashing the beta via https://everythingsmarthome.github.io/everything-presence-one/updating.html still tries to "install" the firmware again, which fails with the above error message.
> 
> Copying the header from /config/esphome/.esphome/packages/***/ directory to /config/esphome/ does resolve the issue.

This fixed it for me as I reference the YAML via ESPHome.

EverythingSmartHome, 2023-08-02:
This is a bit of a tricky one because the firmware from the user guide compiles the header file fine and that works, however when you import the config into the ESPHome add-on (to do OTA updates etc) it won't work because it can't find that file.

I messed around with it a little bit earlier and don't have a great solution yet. There is a couple of ways to download files remotely but didn't get any of them to work yet - "includes" can't include a remote .h file I don't think, remote packages can include remote files but they have to be YAML files, and external components would work but I'd need to transform it into a proper component which will take time (and probably Dev skills I don't have).

I'll keep trying and would be grateful for any suggestions!

pyrodex, 2023-08-02:
Look at https://esphome.io/components/external_components.html for reference on how to do that with packages. You could build your own component package and import those in at build time.

EverythingSmartHome, 2023-08-02:
That's the bit I was referring to above when I said it was probably beyond my skill haha. I can give it a crack though

liquidbrains, 2023-08-03:
Awesome, thanks all.  I was concerned that I missed some documentation somewhere or that I was just being dim.  Now I just need to figure out how to do the multi-zone configuration and what that "Set Distance" button does.

EverythingSmartHome, 2023-08-03:
It's mentioned in the Home Assistant Entities section of the docs

jbrown9100, 2023-10-05:
> Adopting step after flashing the beta via https://everythingsmarthome.github.io/everything-presence-one/updating.html still tries to "install" the firmware again, which fails with the above error [...]

Sorry to ask the nube question here but are you saying you added a copy command into the ESPHome YAML file?  If so what does the syntax look like and where in the YAML does it go. 
If not, are you grabbing the zipped content with the code and doing it locally?  If so how do you then disable the compile an install that happens from ESPHome?

liquidbrains, 2023-10-05:
I actually copied the file to the directory using the Terminal plug-in on
HASS.  The *** part is unknown as it is probably different on your end.
The issue is the header file is not in the correct place to be picked up by
the compiler.

EverythingSmartHome, 2023-10-24:
Please try the new 1.3.0b firmware, hopefully that will resolve it by merging the code from the header file, into the main yaml. Bit of a workaround but I lack the skills (and time I guess) to do a proper custom component and it seems to work OK

liquidbrains, 2023-10-25:
I'm having trouble finding the local git repo to do a git pull so that it'll flash the v1.3.0b firmware.  Do you know if ESPHome changed it recently?

EverythingSmartHome, 2023-10-25:
You should be able to just flash from the user guide here: https://everythingsmarthome.github.io/everything-presence-one/updating.html

liquidbrains, 2023-10-25:
OK, after a _few_ flashes (adopting in ESPHome is not always clear when you have two devices, then changing the name of the wrong device, not figuring out what you did wrong, deleting everything, installing again, realising you installed the non-beta), I've got it working.  No issues with headers.

I've picked up that the LED on the mmWave does not always turn off and I have to reload the config or restart the UART for the light to go out.  I have a log for it, not certain if that'd be useful.

alexhk90, 2023-12-18:
> Please try the new 1.3.0b firmware, hopefully that will resolve it by merging the code from the header file, into the main yaml. Bit of a workaround but I lack the skills (and time I guess) to do a proper custom component and it seems to work OK

This seems to have worked for me - I was able to update to the new 1.3.0b firmware directly from ESPHome in Home Assistant by changing the "package" in the yaml via "Edit" on the device, then "Install" (Wirelessly after restarting the EP1 in "Safe mode" via "Press" in the device page in Home Assistant): 
```
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one-beta.yaml@main
```

Now I just need to work out how to use all the new features ("zones" and "targets"). :)

Ragingfire105, 2024-01-23:
latest beta did not compile for me. 

`fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/kitchen-ep1/.pioenvs/kitchen-ep1/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_f0b0c && [1/2] Building C object CMakeFiles/cmTC_f0b0c.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_f0b0c
    FAILED: cmTC_f0b0c 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_f0b0c.dir/testCCompiler.c.obj  -o cmTC_f0b0c   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)`

alexhk90, 2024-01-23:
I just did the ESPHome 2023.12.9 update for my EP1 with beta firmware (github://everythingsmarthome/presence-one/everything-presence-one-beta.yaml@main) and it seemed to compile and install successfully for me as usual.

EverythingSmartHome, 2024-01-23:
@Ragingfire105 Tried clearing cache?

closed 2024-07-19 by EverythingSmartHome